Zendure/Zendure-HA #1: 5 Doors ???

Wunni73, 2025-02-19:
I installed it but only found 5 doors??? 
are the 5 doors 5 switches??

FireSon, 2025-02-21:
No they were actually test data ;-) I started this integration a week ago with a sample which contained these sensors. 
It is a work in progress, and a somewhat rocky road since the whole pyhon an HA integrations are new to me.

In the current versions the  Zendure API is used to get all the Hyper2000 devices of the account. The values for the sensors are read from the Zendure Cloud MQTT server and shown in HA.

Wunni73, 2025-02-21:
Ich bin auch ein Neuling in Pyhon- und H. Ich erhalte momentan über MQTT-Server die Daten, aber die abfrage ist Grotten schlecht . 
Ich folge dir weiter vielleicht kann ich dir einwenig helfen

closed 2025-02-22 by Wunni73

FireSon, 2025-02-24:
Hi I have updated the integration during the weekend. Maybe you can have a look and see how the performace is...

Wunni73, 2025-02-24:
I'll test the performans live tomorrow.
the switches are still sensors ;-)