Zendure/Zendure-HA #4: the status is not correct

Wunni73, 2025-03-02:
the status is not correct, it is a selection window???

Wunni73, 2025-03-02:
autoModel. Status =

            0: "Nothing",
            6: "Battery priority mode",
            7: "Appointment mode",
            8: "Smart Matching Mode",
            9: "Smart CT Mode",
            10: "Electricity Price",

Wunni73, 2025-03-02:
gridInputPower - no designation  (W)

Wunni73, 2025-03-02:
reverseState is BayPass on/off

FireSon, 2025-03-02:
Nice! I added all your findings in a new version

Wunni73, 2025-03-03:
lampSwitch = Switch

Wunni73, 2025-03-03:
Charging Mode is not the Mode
Pack State is the Mode 1=Charging
                                         0=Sleeping
                                         2=Unloading

FireSon, 2025-03-03:
Thanks again!

Wunni73, 2025-03-04:
the update wasn't good. :-(  All sensors have been recreated :-(

FireSon, 2025-03-04:
Ai sorry for that, I did refactor the integration quite a lot in order to integrate other zendure devices. Also changed the name of the repo again.

closed 2025-03-09 by FireSon