Zendure/Zendure-HA #6: Energy plans and manual values

technoman63, 2025-03-10:
Hello,

Is it possible to add support for different energy plans as well as manual AC input and output control in order to make manual adjustments?

technoman63, 2025-03-13:
[image]

This is the energy plan i want just put this on stop

And 

Put the battery charging with this

[image]
[image]

FireSon, 2025-03-15:
Yes I know the different energyplans. My question is why would you want to use them, if you can create a schedule in Home Assistant en maybe use a Home Assistant integration to get the dynamic prices?

technoman63, 2025-03-15:
I'm in France with a Tempo tariff and I want to:

- Use Zendure's smart mode with my Shelly during the day when I'm not in my energy supplier's off-peak hours (from 6 a.m.) to charge/discharge based on my Shelly consumption and my solar panels.

- From 10 p.m. (off-peak hours), I want to stop all injections, or if the tariff is higher the next day, I want to recharge the battery.

technoman63, 2025-03-16:
Sorry i have two intégration in my home assistant

closed 2025-03-16 by technoman63