Zendure/Zendure-HA #8: Problem output and input power

technoman63, 2025-03-15:
Hi,

I think there's a problem:

- When I charge the battery, I see that the W output is active with values.
- When I discharge the battery, I see that the W input is active with values.

Weren't the two values ​​swapped?

[image]
[image]

Wunni73, 2025-03-15:
these are the original names of Zendure.

FireSon, 2025-03-15:
I just checked to be sure, but yes these are the original names of Zendure. I display them how the Hyper2000 sends them.

Could it be that since the last Firmware update this value has been swapped?

technoman63, 2025-03-15:
I don't know, but I don't find it consistent.

technoman63, 2025-03-16:
Sorry i have two zendure version on my home assistant i'm very sorry

closed 2025-03-16 by technoman63