Zendure/Zendure-HA #9: HACS

pfwolbers, 2025-03-16:
Hello,

Can you make this integration available on HACS?
This makes it easier to install.

Thanks

FireSon, 2025-03-17:
Yes I will try to do that! It will take some time because this is my first custom integration and I have to find out how to do add it to HACS. I think I am halfway now...

pfwolbers, 2025-03-17:
Super, thanks

Mediacj, 2025-03-17:
Maybe I’m not understanding you correctly, but you can already install the integration directly from HACS by adding the following URL as a custom repository: https://github.com/FireSon/Zendure-HA

FireSon, 2025-03-17:
The pullrequest is open for HACS :-)

FireSon, 2025-03-17:
I saw that the list of integrations who want to be added to HACS is very large! This will take an unknown amount of time, so short term this is not a solution.
Therefore I have added a link in teh readme, with which you can install the integration.

Milvus11, 2025-03-19:
Hello, first of all thank you very much for starting such an implementation! Is it normal, that one gets logged out of the Zendure app  when using this integration?

Mediacj, 2025-03-19:
> Hello, first of all thank you very much for starting such an implementation! Is it normal, that one gets logged out of the Zendure app when using this integration?
You have to make a second Zendure login and share your hypers from the priginal login to the second new login. Then both stay activce.

Milvus11, 2025-03-20:
Thank you! I tried to install it yesterday, but somehow it crashed my tibber implementation. I don't know what I did wrong, I reinstalled a Backup from HA. Will try it again on the weekend and then hopefully be able to avoid it or come up with log  files

FireSon, 2025-03-20:
@pfwolbers can I close this issue? It will probably take some time to merge the PR in the HA repository! In the mean time, there is not much more to be done...

closed 2025-03-20 by pfwolbers