Zendure/Zendure-HA #10: Only one Hyper is on line

algernon2607, 2025-03-17:
Hello

i have 2 hyper not in cluster and only one is managed by your intégration : the second is in stand-by and i need  to use the 2400 w output.

Is there a way to use simultaneously the two hyper ??

and where i can find the Acmode control ? It is necessary to do some automatisations to manage my needs.

Thanks for your integration !!

Al

Mediacj, 2025-03-17:
Yes, that would be great. I also have two hypers, and I don’t want them in a cluster but to be able to set them individually to automatic mode or control them via a script with charge or discharge values.

FireSon, 2025-03-17:
I will add a sample automation later, which hopfully explains everything.

bamso, 2025-03-18:
Hi guys, 

If you have a cluster and only one hyper appears...

Remove the grap on zendure application,
Then ,begin zendure intégration's on home assistant you will see all your hyper and batteries

After that you can reconnect to the app and make the cluster. 
It'will be good on zendure app. And you still have more than 1 hyper appears ! 

Good luck 

Samir Bams

algernon2607, 2025-03-19:
It seems that is ok , for this point, with version 1.0.21 .

FireSon, 2025-03-20:
Great to hear that it seems to work properly! Can I close this issue?

closed 2025-03-20 by algernon2607