Zendure/Zendure-HA #11: AC mode directly controllable as selected entity

galliotj, 2025-03-17:
Hi!

First at all, congratulations and thank you very much for the great job done for that new integration! 

Do you think it would be possible to make the "AC mode" parameter editable as a select entity in your integration? Currently, I have my own automation scripts in home assistant using IoBroker and I use a combination of AC mode / Limit input / Limit output entities to do my own smart modes. I tried your manual mode but it is not a reel manual mode because based on a current sensor and the smart maching mode in fact.

Thank you in advance for your feedback.
Julien

FireSon, 2025-03-18:
@galliotj this is an example script to set the manual power with an automation. It checks for the cheapeast energy hours to start charging, the most expensive hours to discharge, and the rest of the day it does smart matching (charging and discharging from solar panels). It is in Dutch, but I think it is quite easy to understand. Also during 'manual power' the integration discharges multiple Hypers evenly.

```
alias: Zendure management
description: ""
triggers:
  - trigger: time_pattern
    seconds: "01"
    minutes: "00"
    hours: "*"
conditions: []
actions:
  - if:
      - condition: state
        entity_id: binary_sensor.goedkoopste_uren_goedkoopste_6_uren_split
        state: "on"
    then:
      - device_id: 9664a678e746046fe741a2738c418935
        domain: select
        entity_id: 28dccf4e2a144aba32df5c499c8c3613
        type: select_option
        option: manual power
      - device_id: 9664a678e746046fe741a2738c418935
        domain: number
        entity_id: 6adfe86cbed9181b861e31908b1e11d8
        type: set_value
        value: -X
    else:
      - if:
          - condition: state
            entity_id: binary_sensor.goedkoopste_uren_3_duurste_uren_split
            state: "on"
        then:
          - device_id: 9664a678e746046fe741a2738c418935
            domain: select
            entity_id: 28dccf4e2a144aba32df5c499c8c3613
            type: select_option
            option: manual power
          - device_id: 9664a678e746046fe741a2738c418935
            domain: number
            entity_id: 6adfe86cbed9181b861e31908b1e11d8
            type: set_value
            value: X
        else:
          - if:
              - condition: and
                conditions:
                  - condition: state
                    entity_id: binary_sensor.goedkoopste_uren_goedkoopste_6_uren_split
                    state: "off"
                  - condition: state
                    entity_id: binary_sensor.goedkoopste_uren_3_duurste_uren_split
                    state: "off"
            then:
              - device_id: 9664a678e746046fe741a2738c418935
                domain: select
                entity_id: 28dccf4e2a144aba32df5c499c8c3613
                type: select_option
                option: smart matching
mode: single
```

closed 2025-03-19 by galliotj

FireSon, 2025-03-19:
@galliotj it is not that I don't want to add it, in fact I will try to add it in the near future. The point is that with the manual operration the current is evenly divided over all hypers, depending on their actual soc. This will keep the maximal availability of all hypers.

FireSon, 2025-03-20:
@galliotj I added this to the latest version (1.0.22) can you check if it works for you as well?

galliotj, 2025-03-21:
Hello! 

Yes it works fine thank you! But one question...why is the "output limit" limited to 200w now? While input limit is still from 0 to 1200W...