Zendure/Zendure-HA #16: Discharge limit

algernon2607, 2025-03-18:
Hello 
i have tested integration with the last firmware.

Smart matching in charge is really good  and i can charge 2400w (1200 X 2 hypers) according to my solar production.

But smart matching in discharge is limited to 1200w max (600 x 2 hypers).

In my country the Inverse Max Power by hyper is 1200w and 2 hypers by phase.

Can you adapt max discharge power to permit max discharge in the regulatory exit limit ???  (1200 by hyper so 2400 w in my case)

FireSon, 2025-03-18:
Per phase you can specify the output limit, what did you specify?

[image]

There are 3 options:
        800 watt output
        1200 watt output, possible circuit overload, use at your own risk!!
        2400 watt output, possible circuit overload, use at your own risk!!
Which one did you select?

algernon2607, 2025-03-18:
Hers is my configuration :
Phase 1   2400w output

Phase 2   800w  useless in my case  (single phase  ..)
Phase 3 800w

Geert2991, 2025-03-18:
[image]

I've also installed the intergration, I like it alot.
But the output slider only let me put out 800watt. Look at the picture above.

FireSon, 2025-03-18:
@algernon2607 can you also check the phase property of your hypers? The Zedure phase check, updates the phase property. If the phase check has not been done, they all report phase 0, which leads to the 2x600 watt discharge. I did not find a way to trigger the phase detection, the Zendure AI told me the phase checks starts automatically when the 'allow cluster' is checked and you restart the Hyper2000.

Geert2991, 2025-03-18:
For me all the fases are 0, soo that will be the problem. Will check if it will change when I let it allow to make a cluster.

Geert2991, 2025-03-18:
Still 800, while the intergration is showing phase 2.

FireSon, 2025-03-18:
What is the Inverse Max Power?

Geert2991, 2025-03-18:
Thats just fine, 1200w

FireSon, 2025-03-18:
Ahh I found it, the matching command had a limitation of 800 watts. I will fix this in the next release!

Geert2991, 2025-03-18:
You're the best! Many thanks!

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

algernon2607, 2025-03-19:
Hello 
Phase 0 and phase check 0 for the 2 hypers ...
 The inverse max power is the regulatory limit impose par country reglementation. It is done when you are configuring  hyper choosing the country.. For my country , 1200 by hyper , 2400 on a single phase.

sankukace, 2025-03-19:
I managed to get 1200W * 2 yesterday when I started the microwave, but it took some time...

The CT mode works very well when there isn’t a big difference, but when the gap is large (oven, heating, etc.), it seems to adjust in steps, and it takes time.

Is it possible to make it faster?

Thanks

algernon2607, 2025-03-19:
@sankukace    I have 2 hyper , no cluster and no smartmeter included in Zendure app .. All is ok with Fireson integration since 1.0.21 except output limit

What is your configuration ??

sankukace, 2025-03-19:
I have 2 Hypers, usually in a cluster, but with the integration, I disable the cluster in the app. I have a Shelly PRO3EM.

algernon2607, 2025-03-19:
@sankukace  ok but how do you obtain  1200W * 2 in output ???

i have also a shelly pro 3em . it is not included in Zendure app because i cannot (do not want also ..) use shelly cloud ( mqtt exclusive)

sankukace, 2025-03-19:
No idea, yesterday when I ran the microwave for 6 minutes, I saw after a while 1200 for each Hyper...

But as FireSon said, he found the issue and will release an update soon. I don't know why it seemed to work yesterday. Or maybe a bug on my end

algernon2607, 2025-03-20:
@FireSon  I reconfigure totaly the integration putting 2400 w in each phase even i am in a single phase.
Phase1 : 2400
phase2 : 2400
phase3 : 2400
it seems work delivering 2400 ( 2 x 1200)  . I confirm the next days with the washing machine.

FireSon, 2025-03-20:
@sankukace  you are right, it is not very responsive, but this will be improved. At this moment I am a bit conservative regarding the number of commands sent to the Hyper2000. I need to find the optimal update rate, and I would not like to 'destroy' and hyper2000 in the proces :-)

bilbon0bud, 2025-03-25:
Yes that good, thanks man.

FireSon, 2025-03-30:
I have added this in version 26. Can you check if it working properly? Unfortunately I cannot do that myself...

closed 2025-04-11 by FireSon