Zendure/Zendure-HA #17: Question: Zendure Current Delta ist always around 40-70 W

Milvus11, 2025-03-20:
Hey, 
thank you for the integration, finally I can use my tibber meter to smart match the Hyper! 
I have a question, my Current Delta Value is always in plus, around 50W. So basically, even though I have enough energy for charging and bypass, I buy 50W kind of permanently. Is there a reason for that? Is it anyhow changeable so that it becomes a net zero?

FireSon, 2025-03-20:
Yes that is correct. I found, by out looking at the Shell3EM communication, that they do that as well. So I thought I might do the same, just to be safe. Once I get some more feedback/experience with these settings I might remove this.

Milvus11, 2025-03-20:
okay, thank you for the fast reply!

Milvus11, 2025-03-20:
just to understand, your answer applies to both questions I wrote, right?

Milvus11, 2025-03-20:
okay, thank you for clarification! I am curious what you will find out. Zendure Forum does not give any Input on that? Would be nice to have a documentation of the shelly implementation for copy and paste

Milvus11, 2025-03-21:
Good morning, 
I am sorry for maybe newbie questions, but I have another: I observed the following this morning: Battery was empty (minimum SocSet), sun started shining, Smart Match mode active. Instead of feeding the energy directly into the grid, to meet the consumption of the flat, the Hyper first charged the battery for around 2-3% and only then the mode switched to feeding the house, solar and battery. Is that intentionally? I work with a flexible energy price (tibber), so in the mornings, the price is typically high, it would make sense for me, to feed the morning sun into my grid instead of the battery. 

Just would like to understand the logic, I guess I anyway need to start building a logic at some point 😅

Milvus11, 2025-03-21:
And now i might have another one: would it make sense to add an adjustable "buffer time" to smart match mode? So if there is a peak in consumption that the smart match doesn't peak in the opposite direction immediately, but waits for 5 seconds and makes a floating mean or so before it reacts? As it keeps that discharge level it sets in that moment for a long time (rn it threw 800 W into the grid, while only 200 necessary bc. There was a spike of something in my meter)

FireSon, 2025-03-30:
@Milvus11 I have updated the update frequency in the latest version (26) 
I understand the question about the flexible energy price. This is easily done using an automation. You can set all the settings via the integration. Below is an example how to do just that. I will close this issue for now.

```
alias: Zendure management
description: ""
triggers:
  - trigger: time_pattern
    seconds: "01"
    minutes: "00"
    hours: "*"
conditions: []
actions:
  - if:
      - condition: state
        entity_id: binary_sensor.goedkoopste_uren_goedkoopste_6_uren_split
        state: "on"
    then:
      - device_id: 9664a678e746046fe741a2738c418935
        domain: select
        entity_id: 28dccf4e2a144aba32df5c499c8c3613
        type: select_option
        option: manual power
      - device_id: 9664a678e746046fe741a2738c418935
        domain: number
        entity_id: 6adfe86cbed9181b861e31908b1e11d8
        type: set_value
        value: -X
    else:
      - if:
          - condition: state
            entity_id: binary_sensor.goedkoopste_uren_3_duurste_uren_split
            state: "on"
        then:
          - device_id: 9664a678e746046fe741a2738c418935
            domain: select
            entity_id: 28dccf4e2a144aba32df5c499c8c3613
            type: select_option
            option: manual power
          - device_id: 9664a678e746046fe741a2738c418935
            domain: number
            entity_id: 6adfe86cbed9181b861e31908b1e11d8
            type: set_value
            value: X
        else:
          - if:
              - condition: and
                conditions:
                  - condition: state
                    entity_id: binary_sensor.goedkoopste_uren_goedkoopste_6_uren_split
                    state: "off"
                  - condition: state
                    entity_id: binary_sensor.goedkoopste_uren_3_duurste_uren_split
                    state: "off"
            then:
              - device_id: 9664a678e746046fe741a2738c418935
                domain: select
                entity_id: 28dccf4e2a144aba32df5c499c8c3613
                type: select_option
                option: smart matching
mode: single
```

closed 2025-03-30 by FireSon

Milvus11, 2025-03-31:
Thank you very much! I'll dive into the automatization. Awesome!