Zendure/Zendure-HA #18: Limit output limited to 200W?

galliotj, 2025-03-21:
Hello,

Thanks a lot for making "AC mode" editable in the last release, it works very well.

But I see that the "output limit" is now limited to 200W...why? While "input limit" is still from 0 to 1200W.

Is it possible to modify maximum output limit to 1200W?

Thank you in advance for your feedback.

FireSon, 2025-03-21:
@galliotj Sorry, my mistake I updated the wrong number. This will be fixed in the next release, probably later today!

galliotj, 2025-03-22:
Thanks a lot for the feedback and the future update! 👍

pacostef57, 2025-03-23:
When I download an install 1.0.22, it still shows the 1.0.21, and the output limit is still 200 W, how is this ?

FireSon, 2025-03-23:
@galliotj  can you try version 23?
@pacostef57 you were right, the version number of 22 was not correct!

sankukace, 2025-03-23:
It's ok, thanks

closed 2025-03-23 by FireSon

galliotj, 2025-03-24:
Works great now, thank you very much !