Zendure/Zendure-HA #19: Flag for Global Server

Maggi0r, 2025-03-22:
I  have a problem to connect to Zendure Server.
My devices are on the Global Server. When i will connect with my credentials i become an Failed to connect Error.
I mean ther Error is, the Integration can not connect to to the Server why they will connect to EU Server.

I hope you have an idea where the Problem is.

Greets Maggi

FireSon, 2025-03-22:
Yes I think that the problem is clear. I will try to fix it. The integration connects to the wrong mqtt server, which is easy to fix.

Maggi0r, 2025-03-22:
Thank you for your work. I wait for the update.

FireSon, 2025-03-22:
Just a quick question, did you change the Zendure host name in the configuration to: https://app.zendure.tech instead of https://app.zendure.tech/eu?

Maggi0r, 2025-03-23:
Yes i have tested this without /eu or with /global. But nothing works.

Greets

FireSon, 2025-03-23:
Can you try it in release 23? I removed the server from the configuration, and automatically try both servers. If it still doesn't work can you create a log file (Enable debug logging) and sent it to me?

Maggi0r, 2025-03-23:
Hello, I tested version 23. But that doesn't work either.
Unfortunately, I can't enable debug logging because I can't even install the integration. I'm posting the logs I found under "Logs for the Integration" here. I hope this helps. Thanks for your work. This are the Logs `Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.config_flow
Quelle: custom_components/zendure_ha/config_flow.py:51
Integration: Zendure Home assistant Integration (Dokumentation, Probleme)
Erstmals aufgetreten: 14:23:24 (4 Vorkommnisse)
Zuletzt protokolliert: 14:39:16

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/config_flow.py", line 51, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/config_flow.py", line 25, in validate_input
    raise ZendureConnectionError
custom_components.zendure_ha.config_flow.ZendureConnectionError: Zendure Integration`

`Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.api
Quelle: custom_components/zendure_ha/api.py:76
Integration: Zendure Home assistant Integration (Dokumentation, Probleme)
Erstmals aufgetreten: 14:23:24 (4 Vorkommnisse)
Zuletzt protokolliert: 14:39:16

<bound method ClientResponse.text of <ClientResponse(https://app.zendure.tech/eu/auth/app/token) [400 Bad Request]> <CIMultiDictProxy('Content-Type': 'application/json;charset=UTF-8', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Date': 'Sun, 23 Mar 2025 13:23:24 GMT', 'Access-Control-Allow-Methods': 'POST, GET, PATCH, DELETE, PUT', 'Access-Control-Max-Age': '3600', 'Access-Control-Allow-Headers': 'Accept-Encoding,Host,Connection,Content-Length,User-Agent,Origin, X-Requested-With, Content-Type, Accept,blade-auth,authorization,Accept-Language,appVersion,captcha-code,captcha-key', 'Access-Control-Allow-Credentials': 'true', 'X-Cache': 'Error from cloudfront', 'Via': '1.1 5f5fdd347d6ea8b242af79ee38a02fae.cloudfront.net (CloudFront)', 'X-Amz-Cf-Pop': 'FRA60-P8', 'X-Amz-Cf-Id': '2Ff0PIM2nAGgB95vuP0YtHDdrh2cCDe1AzMHgOcvcSRCtgv89UvqVQ==')> >
<bound method ClientResponse.text of <ClientResponse(https://app.zendure.tech/eu/auth/app/token) [400 Bad Request]> <CIMultiDictProxy('Content-Type': 'application/json;charset=UTF-8', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Date': 'Sun, 23 Mar 2025 13:23:39 GMT', 'Access-Control-Allow-Methods': 'POST, GET, PATCH, DELETE, PUT', 'Access-Control-Max-Age': '3600', 'Access-Control-Allow-Headers': 'Accept-Encoding,Host,Connection,Content-Length,User-Agent,Origin, X-Requested-With, Content-Type, Accept,blade-auth,authorization,Accept-Language,appVersion,captcha-code,captcha-key', 'Access-Control-Allow-Credentials': 'true', 'X-Cache': 'Error from cloudfront', 'Via': '1.1 5f5fdd347d6ea8b242af79ee38a02fae.cloudfront.net (CloudFront)', 'X-Amz-Cf-Pop': 'FRA60-P8', 'X-Amz-Cf-Id': 'vcFolXWdZ_PrQKOlPMO1BAdbWgxIad9QSt65O5nmrGmWKZI1PZJwoA==')> >
<bound method ClientResponse.text of <ClientResponse(https://app.zendure.tech/eu/auth/app/token) [400 Bad Request]> <CIMultiDictProxy('Content-Type': 'application/json;charset=UTF-8', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Date': 'Sun, 23 Mar 2025 13:26:46 GMT', 'Access-Control-Allow-Methods': 'POST, GET, PATCH, DELETE, PUT', 'Access-Control-Max-Age': '3600', 'Access-Control-Allow-Headers': 'Accept-Encoding,Host,Connection,Content-Length,User-Agent,Origin, X-Requested-With, Content-Type, Accept,blade-auth,authorization,Accept-Language,appVersion,captcha-code,captcha-key', 'Access-Control-Allow-Credentials': 'true', 'X-Cache': 'Error from cloudfront', 'Via': '1.1 376388af58845ad0897ba599cce4d92e.cloudfront.net (CloudFront)', 'X-Amz-Cf-Pop': 'HAM50-C1', 'X-Amz-Cf-Id': '1z-IgFn3ZRZobW8-aFrYYQfAfEpuDUQi22N0ABL9ipVdV1kcHD4fsw==')> >
<bound method ClientResponse.text of <ClientResponse(https://app.zendure.tech/eu/auth/app/token) [400 Bad Request]> <CIMultiDictProxy('Content-Type': 'application/json;charset=UTF-8', 'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Date': 'Sun, 23 Mar 2025 13:39:16 GMT', 'Access-Control-Allow-Methods': 'POST, GET, PATCH, DELETE, PUT', 'Access-Control-Max-Age': '3600', 'Access-Control-Allow-Headers': 'Accept-Encoding,Host,Connection,Content-Length,User-Agent,Origin, X-Requested-With, Content-Type, Accept,blade-auth,authorization,Accept-Language,appVersion,captcha-code,captcha-key', 'Access-Control-Allow-Credentials': 'true', 'X-Cache': 'Error from cloudfront', 'Via': '1.1 e626e3045304034f7160fd4e5f8b6fce.cloudfront.net (CloudFront)', 'X-Amz-Cf-Pop': 'HAM50-C1', 'X-Amz-Cf-Id': 'bhv6_l8rGftFEZh1qlXYePah0xb6Q9dKGSvKP0IL7dZhEMlcPGGAIg==')> >`

`Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.api
Quelle: custom_components/zendure_ha/api.py:75
Integration: Zendure Home assistant Integration (Dokumentation, Probleme)
Erstmals aufgetreten: 14:23:24 (4 Vorkommnisse)
Zuletzt protokolliert: 14:39:16

Authentication failed!`

Greets Maggi

FireSon, 2025-03-23:
Can you test version 24? I looked it up and the v2 part was missing from the url!

Maggi0r, 2025-03-23:
I have tested but them Problem.

Greets Maggi

FireSon, 2025-03-23:
@Maggi0r can you add the same logging again (for  1.0.25)?

Maggi0r, 2025-03-23:
Ok now i became only this one log `Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.zendure_ha.config_flow
Quelle: custom_components/zendure_ha/config_flow.py:51
Integration: Zendure Home assistant Integration (Dokumentation, Probleme)
Erstmals aufgetreten: 20:32:23 (2 Vorkommnisse)
Zuletzt protokolliert: 20:33:08

Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/config_flow.py", line 51, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/config_flow.py", line 25, in validate_input
    raise ZendureConnectionError
custom_components.zendure_ha.config_flow.ZendureConnectionError: Zendure Integration`

Greets Maggi

FireSon, 2025-03-30:
@Maggi0r  can you check this with version 26?

Maggi0r, 2025-03-31:
Hello OK i hae testet and it works with Version 25. But i have now the thame Problems, i be come No Sensor information. All Sensors are unkonwn.

Greets 
maggi

FireSon, 2025-03-31:
@Maggi0r I got feedback from another (global) user that the version pre1.0.17-5 works, can you check that as well?

Maggi0r, 2025-04-01:
OK i Test this at the evening today. Thank you for your Work.

Maggi0r, 2025-04-01:
Hi, I've tested it now. With version 27, the login works, and I can also get data in.
Thanks for that. It would be nice if the battery sensors were also included, such as temperature (sensor packData.serialnumber.maxTemp) and battery SoC (sensor packData.serialnumber.socLevel), as well as voltage (sensor packData.serialnumber.totalVol). Then I'd like to know what this Solarcycle is.

Thanks
Regards, Maggi

closed 2025-04-11 by FireSon