Zendure/Zendure-HA #20: zendure -cluster not manage ?

laurent941, 2025-03-23:
1st thing thank you very much for your work and this native integration on home assistant

I have a problem: I have 3 hyper 2000 in a cluster and I only retrieve the 1  at the top of the list, the other 2 are not retrieved?

do you plan to manage the cluster?
thanks in advance

FireSon, 2025-03-23:
Yes, I am planning to manage the clusters. Apperantly one of the hypers is the 'cluster manager' that is why you only see one. 
Hopefully this week I will be able to figure out the command structure for the cluster.

laurent941, 2025-03-25:
thank you again for your work i am at your disposal for testing 
 thanks to your integration I will be able to compensate for the limitations of the application

STUBBYs1, 2025-04-02:
Vielen Dank für die Top Arbeit und diese native Integration auf Home Assistant

Auch ich habe das Problem: Ich habe 3 Hyper 2000 in einem Cluster und rufe nur den einen aus der Liste ab, die anderen 2 werden nicht abgerufen?

Was macht Ihre Planung der Cluster Verwaltung?
Vielen Dank im Voraus

FireSon, 2025-04-03:
@STUBBYs1 The highest priority at this moment is to get the communication stable. In the mean time I am adding sensors for the Energy Dashboard. Since it is quite easy to remove the cluster (because the integration takes care of the load per phase as well) I will look into that if the other things are working. In a cluster one device is in control, which probably the reason you see only one device. 
Can you try to remove the cluster and see if the devices are there?

STUBBYs1, 2025-04-03:
Hallo FireSon,

Yes i can see 3 Hyper

Mit freundlichen Grüßen

 

<https://github.com/FireSon/Zendure-HA/issues/20#issuecomment-2774775929> ,
TRW3AVCNFSM6AAAAABZS3KNEKVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDONZUG43TK
OJSHE> .
<https://github.com/notifications/beacon/BEZNASJ4DMZMDW6YXTUIO5T2XTRW3A5CNFS
M6AAAAABZS3KNEKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTV

  <https://avatars.githubusercontent.com/u/6523983?s=20&v=4> FireSon left a
comment (FireSon/Zendure-HA#20)

<https://github.com/FireSon/Zendure-HA/issues/20#issuecomment-2774775929> ,
TRW3AVCNFSM6AAAAABZS3KNEKVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDONZUG43TK
OJSHE> .
<https://github.com/notifications/beacon/BEZNASJ4DMZMDW6YXTUIO5T2XTRW3A5CNFS
M6AAAAABZS3KNEKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTV

STUBBYs1, 2025-04-05:
Hallo FireSon,

 

folgende Sachen sind aufgefallen:

1.	Es scheint das dieses HACS Plugin / AddOn für Homeassistant eine
"App" simuliert (in Homeassistant Variante) und dich aus der (normalen)
Zendure App ausloggt.

Man muss sich dort wieder eingeloggt, aber seitdem man sich in der App
wieder eingeloggt hat funktioniert interessanterweise die Steuerung nicht
mehr. 

Also es ist wie ein "Slot" der nur einmalig genommen werden kann.

Ich weiß nicht ob dies in der Dokumentation ggf. steht. ???

Man kann einen Gast Account einrichten via Gerätefreigabe und mit dieser bei
dem HACS Plug In einloggen und siehe da. es geht!

 

2.	Man muss einen zweiten Account nehmen und mit diesem die
Installation teilen. Den nimmt man dann für Homeassistant 
3.	 

Steuerung funktioniert leider aus irgendeinem Grund nicht.

 

Frage:  kann da was verbessert werden ???

 

Mfg

 

<https://github.com/FireSon/Zendure-HA/issues/20#issuecomment-2774775929> ,
<https://github.com/notifications/beacon/BEZNASJ4DMZMDW6YXTUIO5T2XTRW3A5CNFS

<https://github.com/FireSon/Zendure-HA/issues/20#issuecomment-2774775929> ,
<https://github.com/notifications/beacon/BEZNASJ4DMZMDW6YXTUIO5T2XTRW3A5CNFS

STUBBYs1, 2025-04-26:
Hello,
I currently have no values ​​for my three hypermeters for the entities: low_temperature, & _pass.
On the Solar Flow 800, none of the entities are displayed.
Regards

STUBBYs1, 2025-04-26:
Hello FireSon,

I currently have no values ​​for my three Hypers, including the entity: bat_Volt.
Best regards

STUBBYs1, 2025-04-26:
1. After updating to 1.0.40, I noticed that the manual settings automatically reset to their default settings.
2. Furthermore, the settings are difficult to adjust precisely; this could perhaps be improved with a "spin" or "scroll" button.
Best regards

STUBBYs1, 2025-04-26:
Hello FireSon

Other entities are displayed as unknown:
Output Limit, Input Limit, Soc Max, Soc Min, Battery,
Battery Count, Battery Output Power, Battery Input Power, Battery Voltage, Battery Temperature, BuzzerSwitch, EnergyPower, GridInputPower, Main Switch, HeatState, Hub Status, InverseMaxPower, OutputHomePower, OutputHomePowerCycle, SocStatus, SolarInputPower, SolarPower1+2,
Regards

STUBBYs1, 2025-04-26:
Hello FireSon

Is it possible to add additional entities?
For example: Battery SOH, Battery Current Direction, and Battery Rated Capacity.
Regards