Zendure/Zendure-HA #21: Zendure-HA without cloud ?

olivr2s, 2025-03-23:
Hello,
First of all, thank you for this integration into HA, which works well and is regularly updated to become even more comprehensive.

There are ways to bypass the Zendure cloud and use only local MQTT. Are there any plans to use the integration without the cloud?
I've moved my Hyper 2000 off the cloud, to my MQTT broker, and I'm getting the information back. But restructuring the value mappings is a bit more complicated.

FireSon, 2025-03-23:
Yes that is certainly the plan! I did allready some testing with it and that looked very promesing. I also was able to create a sort of hybrid solution where the Zendure App also worked, but the hyper was using a local MQTT server.

olivr2s, 2025-03-23:
This is great news! I will continue to closely monitor future updates/developments.

Andreasq22, 2025-04-07:
Since one of my three hypers is offline (disconnected from cloud) at the moment: I would be very happy about an implementation of local control

masterblaster15, 2025-04-24:
Hallo, erst ein mal vielen Dank für diese Arbeit.
Gibt es Pläne, die Integration ohne die Cloud zu nutzen für den HUB2000?

FireSon, 2025-05-10:
Well the first pre-release with local Mqtt is ready for testing. Please look at the release notes and it would be great to get some feedback in it.

closed 2025-05-10 by FireSon