Zendure/Zendure-HA #22: 1.0.23 zendure manager KO

pacostef57, 2025-03-23:
> @galliotj  can you try version 23?
> @pacostef57 you were right, the version number of 22 was not correct!  

 _Originally posted by @FireSon in [#18](https://github.com/FireSon/Zendure-HA/issues/18#issuecomment-2746134201)_

@FireSon , i tried the 1.0.23... the smart mode is ko. He never start discharge or charge in smart mode.

pacostef57, 2025-03-23:
@FireSon Warning, following this, my Hyper2000 is now showing an error, with error codes 53 and 39... I think it's dead.

FireSon, 2025-03-23:
That does not sound well, sorry to hear that!!
Where did you see these errorcodes? 
Have you tried restarting the hypers?

sankukace, 2025-03-23:
Indeed, I reverted to version 1.0.22 because Smart Matching is acting up... It works fine in 1.0.22.

pacostef57, 2025-03-23:
> En effet, je suis revenu à la version 1.0.22 car Smart Matching fait des siennes... Il fonctionne très bien en 1.0.22.

Comment fait tu pour avoir la 1.0.22, chez moi ça mets la 1.0.21 même quand je selectionne la 1.0.22.

pacostef57, 2025-03-23:
> Ça ne sonne pas bien, désolé de l'apprendre ! Où avez-vous vu ces codes d'erreur ? Avez-vous essayé de redémarrer les hypers ?

I have just one Hyper2000 with one AB2000S, connected to the power outlet. The solar panels are not connected to it; they are independent with their own inverter. The errors appear in the Zendure app. I'm not sure if there's a connection with the fact that your integration managed the Hyper2000 or not. But this is the first time in two months that I've had issues with the Hyper2000. Currently, it seems to be working better; it's not broken, but occasionally I get this error (low voltage) even though I've never had this before, and my network is good.

sankukace, 2025-03-23:
> > En effet, je suis revenu à la version 1.0.22 car Smart Matching fait des siennes... Il fonctionne très bien en 1.0.22.
> 
> Comment fait tu pour avoir la 1.0.22, chez moi ça mets la 1.0.21 même quand je sélectionne la 1.0.22.

Ah oui pardon, j'ai sélectionné 1.0.22 mais cela à mis la 1.0.21.

FireSon, 2025-03-23:
OK clear! I did do some testing to improve the smartmode, appearently I forgot to revert one thing which messed up the smart mode. I fixed it and will release a fixed version soon. Glad to hear that the errors are not releated to the integration.

FireSon, 2025-03-23:
Version 24 should now be available with the fix for the smart mode

pacostef57, 2025-03-23:
> Version 24 should now be available with the fix for the smart mode

I had uninstalled everything and downloaded Zendure Manager 1.0.24 (it displays 1.0.23), but I lost the Hyper2000 device (where you can see all the stats of the Hyper and the AC modes). How did this happen? How can I get it back?

sankukace, 2025-03-23:
No, I no longer have access to the controls. I just reverted to version 1.0.21 and it's working fine again.

sankukace, 2025-03-23:
Phew, great! Everything seems to be working fine again since version 1.0.25.

FireSon, 2025-03-23:
Sorry gens, the api behaved a bit different than expected to get the regional url. In version 25 this should be fixed!

FireSon, 2025-03-23:
In HACS there is an option to 'redownload'

[image]

sankukace, 2025-03-23:
> Sorry gents, the api behaved a bit different than expected to get the regional url. In version 25 this should be fixed!

No worries on my end :) This add-on is already great and keeps getting better every day… :)
Sometimes, you just have to deal with a few teething problems!

pacostef57, 2025-03-23:
> In HACS there is an option to 'redownload'
> 
> [image]

It seems to me that I had tried and only the Zendure Manager appeared, not the Hyper2000... I will check that again.

closed 2025-03-26 by pacostef57