Zendure/Zendure-HA #23: Request for individual battery information

sankukace, 2025-03-23:
Would it be possible to add individual battery information to your add-on?
I have two Hyper units with two batteries each, and with ioBroker I can retrieve the charge percentage of each battery individually.

The MQTT information is in packData.

Thanks in advance, and great job on the add-on! I'm almost at the point where I can stop using ioBroker.
I love how it's evolving, and I always look forward to installing the new versions :)

Example : 

[image]

FireSon, 2025-03-23:
Yes that is not very difficult. How does iobroker handle the 1-4 batteries? Are they seperate devices? Or just a sensor for the charge level sufficient per battery?

sankukace, 2025-03-23:
[image]

"Personally, I only need the fill percentage, but here is all the available information. Better to have too much than too little — it might be useful to others."

sankukace, 2025-03-29:
Hello,
I’m looking forward to receiving this information so I can move away from ioBroker.

The pre-1.0.26 version doesn’t have it, and the current sensor no longer works, so I went back to version 1.0.25.

sankukace, 2025-04-01:
Here are the MQTT topics on zendure.eu:

zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/model  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/socLevel  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/maxTemp  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/minVol  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/maxVol  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/totalVol  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******9/sn

In my case, the following are especially relevant:

zendure-solarflow/0/*******/eW0EqtSH/packData/CO4FHMG4*******/socLevel  
zendure-solarflow/0/*******/200U250e/packData/CO4HMCJMD*******/socLevel

I believe 200U250e refers to Hyper #1, and eW0EqtSH to Hyper #2.

Thanks again in advance!

FireSon, 2025-04-01:
@sankukace There are multiple ways to do this. We can add 'new devices' for each battery, which is straightforward and all the properties can easily be added to the device (battery). The Hyper will get max. 4 properties with the serial number of the battery.

The other option is to add all the battery properties to the Hyper2000, with the prefix of bat1..bat4. Since there are many properties this might get a bit confusing. What is your preference?

sankukace, 2025-04-12:
Yes:) 

Thanks :)

closed 2025-04-12 by sankukace