Zendure/Zendure-HA #25: Upgrade to 1.0.25 -> failed to connect

Popoboxxo, 2025-03-23:
Firstly, this is a really cool project, and I appreciate all the hard work that's gone into it!

Regarding a current bug: since updating from version 1.0.23 to 1.0.25, I'm unable to log in with my EU account. I'm receiving a connection error."

[image]

Popoboxxo, 2025-03-23:
By the way, the screenshot is from the re-setup after I tried to set up the integration again due to an error. The username and password are right.

FireSon, 2025-03-23:
@Popoboxxo well that is strange, I also have an eu account, and that works! Can you delete the entry and re-add it again?

Popoboxxo, 2025-03-23:
Sure, I will. Are there any config files at the file level that I should delete?

FireSon, 2025-03-23:
No just delete the configuration is enough

Popoboxxo, 2025-03-23:
Still get an error.
[image]

The password does not contains special Chars, but the mails have an "+" init. 
Any validation Errors?  

" info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^"

Popoboxxo, 2025-03-23:
Jop.

Popoboxxo, 2025-03-23:
I rolled back to version 1.0.23 earlier, and I was able to get the integration working again (without re-creating it) and access all the entities."

FireSon, 2025-03-30:
@Popoboxxo can you check this in version 26?

Popoboxxo, 2025-03-30:
Still getting an Error.

[image]

And:

Logger: homeassistant.config_entries
Quelle: config_entries.py:753
Erstmals aufgetreten: 01:35:31 (65 Vorkommnisse)
Zuletzt protokolliert: 01:46:59

Error setting up entry Zendure Integration -

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 753, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/binary_sensor/__init__.py", line 139, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 182, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry Zendure Integration - my mail (KEY) for zendure_ha.binary_sensor has already been setup!

I'll try to deleted it tomorrow and Install it again.

FireSon, 2025-03-31:
Can you check the pre-release 27? You have to select that manually in HACS

FireSon, 2025-03-31:
Yes, if you turn on the debug logging for the integration, you can see much more info under: Settings\System\Log if you select "Show Raw Logs". It is important to know what url is selected for the api and for mqtt

Popoboxxo, 2025-03-31:
Debug-log: 

[36m2025-03-31 14:33:06.700 DEBUG (MainThread) [custom_components.zendure_ha.config_flow] Check API connection[0m
[32m2025-03-31 14:33:06.701 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure[0m
[31m2025-03-31 14:33:07.311 ERROR (MainThread) [custom_components.zendure_ha.api] Unable to connect to Zendure ![0m
[31m2025-03-31 14:33:07.311 ERROR (MainThread) [custom_components.zendure_ha.config_flow] Unexpected exception
Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/config_flow.py", line 51, in async_step_user
    info = await validate_input(self.hass, user_input)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/config_flow.py", line 25, in validate_input
    raise ZendureConnectionError
custom_components.zendure_ha.config_flow.ZendureConnectionError: Zendure Integration[0m

I think i got it... Its an Input-String Error.

My "Second-Zendure-Account" Contains an "+" in the email. I think this is the "Validation Input Error". 

Because:

[image]

It is working with my Main-Mail-Account.

And the interpretation of the Input Validation has changed since 1.0.23.

Popoboxxo, 2025-03-31:
Very strange, I can no longer log into the Zendure account with the „+“ in the mail in the app either. But your integration in version 23 still manages it :D....

Popoboxxo, 2025-03-31:
Thanks for looking into the case anyway!