Zendure/Zendure-HA #28: smart matching and soc limit

algernon2607, 2025-03-26:
Hello @FireSon 

Your smart matching mode is really good.
But when a hyper stops after reaching the SOC limit (in charge or discharge), the zendure manager does not manage the setpoint to transfer power to the second hyper.
So , there is no more smart matching (only the half ).
Can this be fixed?

FireSon, 2025-03-26:
Yes certainly, currently working on that!

algernon2607, 2025-03-26:
Great news !!! Thanks you !!!

FireSon, 2025-03-30:
Can you check this in the new version 26?

algernon2607, 2025-03-31:
1.0.26 is NOK . zendure current delta and zendure current power are unavaible .
I  return to 1.0.25

sankukace, 2025-03-31:
Hi, idem, zendure current delta and zendure current power are unavaible

FireSon, 2025-04-11:
Should be fixed in version 29, if you find a problem in that version please reopen this issue!

closed 2025-04-11 by FireSon