Zendure/Zendure-HA #29: What can I control?

phlexss, 2025-03-27:
First of all many thanks for all the work being put into this! I hope you can make a fully functional local api in the end!

I've been using the integration for a short time now and I can get all the values from the Hyper2000 just fine but I'm wondering what I can actually change on the H2000 from HA?

Can I use the Zendure Power Manager when I have just one H2000? I would love to be able to turn Smart Mode On and Off because it seems when Smart Mode is On and the battery is below the "Stop Discharging" threshold it doesn't automatically recharge if the house-demand is more then solar can provide.....it just goes into Bypass and redirects all solar to the house but the battery slowly keeps draining without being recharged....dropping way below the "Stop Discharging" threshold. So I would like to disable Smart Mode from HA so it will recharge from the available solar.....or even better: recharge from AC.

That's my other question: the "AC mode" setting doesn't seem to change anything on the H2000. If I set this to "Output mode" it doesn't go to output mode on the H2000 and the sliders for Limit Input and Limit Output seem to be read-only: changing them doesn't change anything on my H2000 and when I leave the configuration in HA and go back to it they are back to whatever is configure in the Zendure app.

Any help on how I should use the integration and what I can change is greatly appreciated!

phlexss, 2025-03-27:
Ok....found out that the Zendure Manager is working perfectly.....even better than the official smart mode.....because now it actually automatically recharges when the battery is below the lower threshold! 

Thought I already tried this yesterday.....don't know why it didn't seem to work then. Anyway.....seems to work perfectly for now and I'll keep an eye on it....

Thanks again for all your efforts!

phlexss, 2025-03-27:
Sorry about my activity in this ticket but I'm still trying to understand how the Zendure Hyper2000 works.

Regarding automatic recharging when dropping below the "Stop Discharging" threshold; it actually seems that this integration's Smart Mode works the same as the official Smart Mode.....it was just that earlier today my H2000 suddenly dropped to almost 0% and then it started recharging using AC.

Now the discharge limit is set to 15%......the actual charge is 11% and when the demand of the house exceeds the solar-input it doesn't charge the battery but directs all solar to the house. So if solar input never exceeds the house demand the battery charge  will continue to drop to almost 0% before the "emergency recharge" kicks in again and recharges the battery (to the discharge-limit?).

Is the above description how it should be? To me it would be more logical if the H2000 would give priority to recharging to the discharge threshold whenever it drops below that because I'd rather not let the battery discharge to 0%......

Any insghts are appreciated!

FireSon, 2025-03-28:
@phlexss the behaviour you described is seen by other users as well. It is probably not releated to the HA integration. Sometimes it could have to do with the battery callibration. Then it just loads to 100% and discarges to 0% in order to know the limits. 
Hope that helps!

FireSon, 2025-03-30:
In version 26, I added some additional checks that a device is not used outside the operating range. Also the firmware should take care of that. I will close this ticket for now, but feel free to open it again if you have still problems!

closed 2025-03-30 by FireSon