Zendure/Zendure-HA #31: integration into the Energy Board

Wunni73, 2025-03-29:
Can you generate the data so that I can integrate it into the energy board (solar and battery)

FireSon, 2025-03-29:
Sure, can you specify how the data needs to be generated?

Wunni73, 2025-03-29:
https://www.home-assistant.io/docs/energy/

FireSon, 2025-03-30:
Thanks I will look at it!

Wunni73, 2025-03-30:
https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities

sankukace, 2025-04-01:
In MQTT, it's the topic:

zendure-solarflow/0/****/eW0EqtSH/calculations
zendure-solarflow/0/****/eW0EqtSH/calculations/gridInputEnergyTodayWh
zendure-solarflow/0/****/eW0EqtSH/calculations/gridInputEnergyTodaykWh

...

[image]

With this, I think we have everything we need for the energy dashboard.

GeneralRurra, 2025-04-07:
Have you already found something to query? @FireSon

SleXx88, 2025-04-07:
With the standard Energy Dashboard in Home Assistant, it is necessary to use net amounts. Of course it is enough to calculate the AC kWh generated, but I also think it makes sense to include the losses.I will write a separate YAML about this and then publish it here.

GeneralRurra, 2025-04-07:
Ok, because unfortunately the values don't fit when I have them calculated 

[image]

[image]

[image]

FireSon, 2025-04-19:
Should be available in version 36, so closing it for now

closed 2025-04-19 by FireSon