Zendure/Zendure-HA #32: No data retrieved – Sensor registration error with zendure_ha HACS integration

GeneralRurra, 2025-03-29:
I installed the zendure_ha integration via HACS and followed the setup instructions. However, no data is being retrieved – regardless of whether I use my main account or a secondary account (which has proper access to my system).

When checking the logs, I’m seeing the following error related to duplicate sensor registration:

```
Logger: homeassistant.components.sensor
Quelle: helpers/entity_platform.py:633
Integration: Sensor ([Dokumentation](https://www.home-assistant.io/integrations/sensor), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Erstmals aufgetreten: 21:05:54 (4 Vorkommnisse)
Zuletzt protokolliert: 21:18:24

Error adding entity sensor.zendure_current_power for domain sensor with platform zendure_ha
Error adding entity sensor.zendure_current_delta for domain sensor with platform zendure_ha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 633, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 770, in _async_add_entity
    entity.add_to_platform_start(
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        self.hass,
        ^^^^^^^^^^
        self,
        ^^^^^
        self._get_parallel_updates_semaphore(hasattr(entity, "update")),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 194, in add_to_platform_start
    super().add_to_platform_start(hass, platform, parallel_updates)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1352, in add_to_platform_start
    raise HomeAssistantError(
    ...<2 lines>...
    )
homeassistant.exceptions.HomeAssistantError: Entity 'sensor.zendure_current_power' cannot be added a second time to an entity platform
``` 

This applies to at least:

sensor.zendure_current_power

sensor.zendure_current_delta

Even after manually deleting the affected entities and restarting Home Assistant, the error reappears and no data is shown.

System Information:

- Device: Hyper 2000

- Modules: 2x AB2000

- Server: Global Server

- Home Assistant Version: (Home Assistant Core 2025.3.4 )

- Integration: zendure_ha (installed via HACS)

What I’ve tried:

- Logging in with both main and secondary accounts

- Deleting the affected sensors manually

- Restarting Home Assistant

- Reinstalling the integration via HACS

Expected behavior:
Sensors should be added once and start retrieving data.

Actual behavior:
Sensors are not added properly, and no data is retrieved. The log shows an error about duplicate sensor registration.

Steps to reproduce:

1. Install the integration via HACS

2. Set up using valid account credentials

3. Restart Home Assistant

4. Observe the log – error appears, no data is shown

[image]
[image]

FireSon, 2025-03-30:
Thanks for the detailed report! Since the integration adds all sensors automatically, it happens during a restart that the described error occurs, when the same sensor is added the second time. This error actually can be ignored, I will try to remove it in the future.

A few people have had 'startup' problems, most of the time I ask them to remove the configuration and add it again. There were several where it all of a sudden seemed to work. It not would it be possible to enable 'debug logging' for the integration and look if there are any clues in there?

GeneralRurra, 2025-03-30:
Here is a modified file; I have censored my ID, battery, and email.

If you’d like, feel free to upload or paste it here — I’ll take a look and help you with whatever you need.

What exactly would you like to do or check with it?

[home-assistant_zendure_ha_2025-03-30T14-29-49.561Z.log](https://github.com/user-attachments/files/19526115/home-assistant_zendure_ha_2025-03-30T14-29-49.561Z.log)

FireSon, 2025-03-30:
Yes that is understandable and wise! The interesting part is this:
```
2025-03-30 16:29:38.794 INFO (paho-mqtt-client-42089:171f5481-839f-4bdb-97db-7cdb8b0b3ca6) [custom_components.zendure_ha.api] Client has been connected
2025-03-30 16:29:38.795 INFO (paho-mqtt-client-42089:171f5481-839f-4bdb-97db-7cdb8b0b3ca6) [custom_components.zendure_ha.api] Client has been disconnected
```

It seems that the mqtt connection is closed immdiately after connecting, resulting in (obviously) no updated values. It also reads that the url is https://app.zendure.tech/as which I don't really know if that exists. Also from the api the used mqtt broker is read (not visible in the logfile) and if this address is not correct, of coarse the connection will close.

I will add the mqtt address to the logfile, and do you know which Zendure cloud sever you are using? Are you in europe? (Since I saw german texts ;)

GeneralRurra, 2025-03-30:
I am on the global server

FireSon, 2025-03-30:
Can you test this with version 26? If it doesn't work can you send me the debug logging to check which server is used for the connection?

GeneralRurra, 2025-03-30:
```

[homeassistant.loader] We found a custom integration zendure_ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-03-31 00:49:18.020 WARNING (SyncWorker_0) 

[custom_components.zendure_ha.zendurephase] Create phase 1 with 3600 watt output and 3000 watt input 3000 watt output, possible circuit overload, use at your own risk!!
2025-03-31 00:57:21.599 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 2 with 3600 watt output and 3000 watt input 3000 watt output, possible circuit overload, use at your own risk!!
2025-03-31 00:57:21.599 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 3 with 3600 watt output and 3000 watt input 3000 watt output, possible circuit overload, use at your own risk!!
2025-03-31 00:57:21.600 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-03-31 00:57:21.605 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-03-31 00:57:21.605 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-03-31 00:57:21.887 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/as => Mqtt: mq.zen-iot.com
2025-03-31 00:57:21.887 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-03-31 00:57:22.319 INFO (MainThread) [custom_components.zendure_ha.api] Getting device details for [60532] ...
2025-03-31 00:57:22.875 INFO (MainThread) [custom_components.zendure_ha.api] Got data for [60532] 4...
2025-03-31 00:57:22.875 INFO (MainThread) [custom_components.zendure_ha.api] Adding device: ck5R1F59 Hyper 2000
2025-03-31 00:57:22.876 INFO (MainThread) [custom_components.zendure_ha.api] Data: {'isShare': False, 'productKey': 'gDa3tb', 'productModelId': 41, 'productName': 'Hyper 2000', 'deviceName': 'Hyper 2000', 'deviceId': 60532, 'deviceKey': 'ck5R1F59', 'snNumber': 'EE1LYMFEM270992', 'onlineFlag': '00601', 'standard': '', 'consumerId': 42089, 'shortCode': 'E1', 'productType': 17, 'priceType': '', 'id': 14266, 'masterSwitch': True, 'restState': 0, 'restTime': '', 'hubState': 0, 'wifiState': True, 'wifiName': 'FRITZ!Repeater 2400', 'mac': '94:c9:60:c7:7c:46', 'ip': '192.168.178.175', 'buzzerSwitch': False, 'autoHeat': 0, 'loraInvState': 0, 'loraAddr': '', 'ambientSwitch': False, 'ambientLightMode': 0, 'ambientLightColor': 65280, 'ambientLightNess': 20, 'solarInputPower': 0, 'pvPower1': 0, 'pvPower2': 0, 'pvPower3': 0, 'pvPower4': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'outputInversePower': 0, 'bindSn': '', 'invOutputPower': 0, 'outputLimit': 0, 'inputLimit': 144, 'remainOutTime': 14400, 'remainInputTime': 59940, 'socSet': 1000, 'minSoc': 300, 'packState': 0, 'packNum': 2, 'electricLevel': 30, 'isDeleted': 0, 'createTime': '2024-08-19T12:57:07Z', 'updateTime': '2025-03-30T22:57:09Z', 'inverseMaxPower': 1200, 'pvBrand': 0, 'brandId': 0, 'cycle': 0, 'blueOta': 1, 'solarInputPowerCycle': 0, 'solarInputPower1Cycle': 0, 'solarInputPower2Cycle': 0, 'packInputPowerCycle': 0, 'outputPackPowerCycle': 0, 'outputInverseCycle': 0, 'outputHomePowerCycle': 0, 'gridInputPowerCycle': 0, 'gridInputPower': 0, 'smartMode': 0, 'smartPower': 0, 'pass': 0, 'lowTemperature': 0, 'heatState': 0, 'modelName': 'Hoymiles-HM-800', 'passMode': 0, 'lowTempNotice': 0, 'autoRecover': 0, 'acMode': 1, 'acDelay': 0, 'acSwitch': 0, 'dcSwitch': 0, 'acOutputPower': 0, 'dcOutputPower': 0, 'acOutputPowerCycle': 0, 'dcOutputPowerCycle': 0, 'socStatus': 0, 'gridStandard': '0', 'phaseCheck': 0, 'priority': 0, 'gridReverse': 1, 'reverseState': 0, 'energyPower': 0, 'lampSwitch': 1, 'clusterSw': 0, 'hyperTmp': 0, 'plugState': 0, 'gridOffMode': 0, 'localState': 1, 'panelType': 0, 'panelPower': 0, 'panelNums': 0, 'fanSwitch': 0, 'fanSpeed': 0, 'outputPower': 0, 'gridState': 0, 'chargePower': 0, 'faultNum': 0, 'priceUpdateDate': '', 'gridOffPower': 0, 'socLimit': 0, 'signUrl': '', 'signCode': '', 'aiSet': '', 'aiState': 0, 'packDataList': [{'createTime': '2024-09-30T03:14:53Z', 'updateUser': 0, 'updateTime': '2025-03-30T22:57:22.75Z', 'isDeleted': 0, 'id': 684152, 'deviceId': 60532, 'consumerId': 42089, 'socLevel': 29, 'state': 0, 'maxTemp': 290.1, 'batteryCode': 'CO4HMDGME504152', 'softVersion': 4113, 'version': '1.0.17', 'sn': 'CO4HMDGME504152', 'heatState': 0, 'packType': 2, 'packName': '', 'packModel': ''}, {'createTime': '2024-09-30T03:18:14Z', 'updateUser': 0, 'updateTime': '2025-03-30T22:57:22.75Z', 'isDeleted': 0, 'id': 685085, 'deviceId': 60532, 'consumerId': 42089, 'socLevel': 32, 'state': 0, 'maxTemp': 286.1, 'batteryCode': 'CO4HMC4MDA03386', 'softVersion': 4113, 'version': '1.0.17', 'sn': 'CO4HMC4MDA03386', 'heatState': 0, 'packType': 2, 'packName': '', 'packModel': ''}], 'todayEnergy': 0, 'temperatureUnit': 0, 'faultFlag': False, 'bateUser': False, 'deviceId2': 60532, 'autoModel': 9, 'solarPower1': 0, 'solarPower2': 0, 'bindProduct': None, 'bindName': '', 'bindDeviceKey': '', 'bindDeviceState': '', 'bindId': None, 'aceSolarPower': 0, 'clusterId': None, 'nodeStatus': None, 'clusterStatus': None, 'hyperNum': 1, 'hemsId': None, 'hemsShareFlag': False, 'remindTime': '', 'networkType': 0, 'energyDate': '2025-03-30', 'otaBatch': False, 'hemsName': ''}
2025-03-31 00:57:22.876 INFO (MainThread) [custom_components.zendure_ha.api] Getting device details for [60534] ...
[... cut ...]
2025-03-31 00:57:24.244 INFO (paho-mqtt-client-42089:2daba21e-d7c0-4d7f-9812-5590c893a3bc) [custom_components.zendure_ha.api] Client has been disconnected; trying to restart
2025-03-31 00:57:25.876 INFO (paho-mqtt-client-42089:2daba21e-d7c0-4d7f-9812-5590c893a3bc) [custom_components.zendure_ha.api] Client has been connected
2025-03-31 00:57:25.878 INFO (paho-mqtt-client-42089:2daba21e-d7c0-4d7f-9812-5590c893a3bc) [custom_components.zendure_ha.api] Client has been disconnected; trying to restart

```

FireSon, 2025-03-31:
Ahh thanks again for the complete report! The mqtt address is part of the api response from Zendure. I had hoped to use this address, since I do not have to 'hardcode' it myself. 
Can you check the pre-release 27? You have to select that manually in HACS

GeneralRurra, 2025-03-31:
i only have it on my cell phone and i was logged out 

and I have stopped my mqtt requests that I have queried so far

FireSon, 2025-03-31:
Ahh thanks for the update, I just remembered that for the global mqtt server there is a different setting to be used. I have changed that one and created a new pre release v1.0.27-4 can you test that one?

GeneralRurra, 2025-03-31:
I'll do it right away I'm digging through your code :) I have to save my data that I'm trying to save

FireSon, 2025-03-31:
It is possible that the mqtt url was correct after all...

FireSon, 2025-03-31:
OK, I adjusted the mqtt url, can you check pre1.0.27-5?

GeneralRurra, 2025-03-31:
```
2025-03-31 19:14:36.506 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration zendure_ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2025-03-31 19:19:11.650 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 outputHomePower => 22 old:21
2025-03-31 19:19:12.685 WARNING 
2025-03-31 19:19:12.688 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarInputPower => 40 old:41
2025-03-31 19:19:12.695 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 outputHomePower => 21 old:22
2025-03-31 19:19:12.696 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarPower1 => 21 old:22

2025-03-31 19:19:13.546 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 1 with 2400 watt output and 800 watt input 800 watt output
2025-03-31 19:19:13.546 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 2 with 2400 watt output and 800 watt input 800 watt output
2025-03-31 19:19:13.546 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 3 with 2400 watt output and 800 watt input 800 watt output
2025-03-31 19:19:13.546 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-03-31 19:19:13.547 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-03-31 19:19:13.548 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure

2025-03-31 19:19:13.903 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarInputPower => 41 old:40
2025-03-31 19:19:13.904 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 outputHomePower => 19 old:21
2025-03-31 19:19:13.904 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarPower1 => 22 old:21

2025-03-31 19:19:13.943 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/v2 => Mqtt: mq.zen-iot.com
2025-03-31 19:19:13.944 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-03-31 19:19:14.382 INFO (MainThread) [custom_components.zendure_ha.api] Getting device details for [60532] ...
2025-03-31 19:19:14.740 INFO (MainThread) [custom_components.zendure_ha.api] Got data for [60532] 4...
2025-03-31 19:19:14.741 INFO (MainThread) [custom_components.zendure_ha.api] Adding device: ck5R1F59 Hyper 2000
2025-03-31 19:19:14.741 INFO (MainThread) [custom_components.zendure_ha.api] Data: {'isShare': False, 'productKey': 'gDa3tb', 'productModelId': 41, 'productName': 'Hyper 2000', 'deviceName': 'Hyper 2000', 'deviceId': 60532, 'deviceKey': 'ck5R1F59', 'snNumber': 'EE1LYMFEM270992', 'onlineFlag': '00601', 'standard': '', 'consumerId': 42089, 'shortCode': 'E1', 'productType': 17, 'priceType': '', 'id': 14266, 'masterSwitch': True, 'restState': 0, 'restTime': '', 'hubState': 0, 'wifiState': True, 'wifiName': 'FRITZ!Repeater 2400', 'mac': '94:c9:60:c7:7c:46', 'ip': '192.168.178.175', 'buzzerSwitch': False, 'autoHeat': 0, 'loraInvState': 0, 'loraAddr': '', 'ambientSwitch': False, 'ambientLightMode': 0, 'ambientLightColor': 65280, 'ambientLightNess': 20, 'solarInputPower': 42, 'pvPower1': 22, 'pvPower2': 20, 'pvPower3': 0, 'pvPower4': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 20, 'outputInversePower': 0, 'bindSn': '', 'invOutputPower': 0, 'outputLimit': 0, 'inputLimit': 144, 'remainOutTime': 59940, 'remainInputTime': 59940, 'socSet': 1000, 'minSoc': 300, 'packState': 0, 'packNum': 2, 'electricLevel': 30, 'isDeleted': 0, 'createTime': '2024-08-19T12:57:07Z', 'updateTime': '2025-03-31T17:18:49Z', 'inverseMaxPower': 1200, 'pvBrand': 0, 'brandId': 0, 'cycle': 0, 'blueOta': 1, 'solarInputPowerCycle': 0, 'solarInputPower1Cycle': 0, 'solarInputPower2Cycle': 0, 'packInputPowerCycle': 0, 'outputPackPowerCycle': 0, 'outputInverseCycle': 0, 'outputHomePowerCycle': 0, 'gridInputPowerCycle': 0, 'gridInputPower': 0, 'smartMode': 0, 'smartPower': 0, 'pass': 2, 'lowTemperature': 0, 'heatState': 0, 'modelName': 'Hoymiles-HM-800', 'passMode': 0, 'lowTempNotice': 0, 'autoRecover': 0, 'acMode': 1, 'acDelay': 0, 'acSwitch': 0, 'dcSwitch': 0, 'acOutputPower': 0, 'dcOutputPower': 0, 'acOutputPowerCycle': 0, 'dcOutputPowerCycle': 0, 'socStatus': 0, 'gridStandard': '0', 'phaseCheck': 0, 'priority': 0, 'gridReverse': 1, 'reverseState': 0, 'energyPower': 0, 'lampSwitch': 1, 'clusterSw': 0, 'hyperTmp': 0, 'plugState': 0, 'gridOffMode': 0, 'localState': 1, 'panelType': 0, 'panelPower': 0, 'panelNums': 0, 'fanSwitch': 0, 'fanSpeed': 0, 'outputPower': 0, 'gridState': 0, 'chargePower': 0, 'faultNum': 0, 'priceUpdateDate': '', 'gridOffPower': 0, 'socLimit': 0, 'signUrl': '', 'signCode': '', 'aiSet': '', 'aiState': 0, 'packDataList': [{'createTime': '2024-09-30T03:14:53Z', 'updateUser': 0, 'updateTime': '2025-03-31T17:19:14.618Z', 'isDeleted': 0, 'id': 684152, 'deviceId': 60532, 'consumerId': 42089, 'socLevel': 30, 'state': 0, 'maxTemp': 296.1, 'b [... cut ...]
2025-03-31 19:21:07.576 INFO (paho-mqtt-client-42089:82a5c4b0-059c-4687-a9cd-3d0ef227e159) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarPower2 => 15 old:16
2025-03-31 19:21:07.593 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarPower1 => 18 old:17
2025-03-31 19:21:07.594 INFO (paho-mqtt-client-42089:73f58d51-a6c2-4b84-8adc-e48e63453cba) [custom_components.zendure_ha.sensor] State: Hyper 2000 solarPower2 => 15 old:16

```
now i get data 

[image]
[image]
[image]

closed 2025-03-31 by FireSon