Zendure/Zendure-HA #35: 1.0.26 NOK

algernon2607, 2025-03-31:
Sensors powers are unaivalables ( zendure  cuurent delta and zendure cuurent power ).
Back to 1.0.25

phlexss, 2025-03-31:
I have the same problem.

algernon2607, 2025-03-31:
i'gonna to try but I must wait for the sun

phlexss, 2025-03-31:
I  installed 1.0.26 again....seems like smart mode IS working.

....didn't expect de delta- and power sensors to be gone so I thought the integration had a problem. The other sensors seem to work fine and seem to be updating much more frequent.

I assume these sensors get their info through the cloud? Isn't there a potential of "overloading" the Zendure servers if too many people start using this and the update frequency is this high?

What is the "Scan Interval" setting for in the overall integration configuration? This is set to 90 seconds.

Any way.....thanks again for developing this integration!

algernon2607, 2025-03-31:
In my case , nothing is done:
manual power : nothing when handle  number zendure manual power in charge or discharge
smart matching : nothing happens 

sensors seems ok 
zendure current delta and zendure current power are helpful to monitor the zendure

i must charge so back to 1.0.25
ready to do some try again ..

algernon2607, 2025-03-31:
In fact Automodel is set to nothing ...

FireSon, 2025-03-31:
There was a problem with charging, can you check the pre-release 27? You have to select that manually in HACS

algernon2607, 2025-03-31:
My first feeback:
 mode OFF : nothing , the charge:discharge go on
manual power : it seems ok , but just a small try . Perhaps , in the afternoom
smart matching :  ok  but the smartmeter is not zero but -50w almost.
I go on with 1.0.27 release2 to give you more informations.

FireSon, 2025-03-31:
OK, thanks, pre1.0.27-3 is out now, which fixes a small issue with none-hyper2000 devices. 
OFF only changes something if you go from manual/smart mode back to OFF. Not sure if you did just that. In OFF mode the Zendure manager does not do anything.

The -50w is correct, has to do with the minimum power which is needed for charging.

pacostef57, 2025-03-31:
Fireson, did you know that in smart mode, the Zendure Manager has never been stable for more than a few hours at my place. There is almost always a freeze that ends up happening. I then have to disable the integration, go back to the Zendure app, stop everything, and then restart a mode from the Zendure app. I now know several people from the French HACF forum who are experiencing the same issue. It's very annoying to constantly find yourself with a smart charge or discharge mode stuck in the mode.

FireSon, 2025-03-31:
@pacostef57 sorry to hear that! When this happens, can you enable the debug logging, and maybe share the logfile with me? Some people have reported similar problems to me, but it is very difficult to pinpoint the problem! 

In the last updated I have tried to improve the communication, allowing the Hyper2000 more time to adjust to the new setpoint. I have asked Zendure for more information on it, and they supplied me with more details which I used in the latest release. 

 Any information on the subject is much approciated!

Popoboxxo, 2025-03-31:
I am observing the same behavior

pacostef57, 2025-03-31:
I tested the smart mode, then I turned it off, but it remains stuck in the state (discharging in my case). I have to go and force an AC charging mode in the Zendure app to stop the issue.
Cela ne marche toujours pas.

FireSon, 2025-03-31:
@pacostef57 Thanks for the feedback, I now set the limits to 0 if you select 'None'
This will be available in version 1.0.27

pacostef57, 2025-03-31:
Ok, thx. But the main issue remains the freezing of the smart mode. It has been present since the earliest versions.
I should specify that everything works perfectly with the Zendure app."

FireSon, 2025-03-31:
Well like I said previously, I did some changes to hopefully eliminate these problems. They are very difficult to find so if you have any pointers please let me know! For now I will close this issue, since the initial issue is solved.

closed 2025-03-31 by FireSon