Zendure/Zendure-HA #37: Battery temperature for AB2000 not available

SleXx88, 2025-04-03:
Hi,
is it possible to add the battery temperature from the AB2000?
In my case, in use with the SF800.
Many thanks

Maggi0r, 2025-04-03:
Das habe ich auch schon angeregt ob es möglich ist diesen und auch den Spannungssensor abzufragen.
Da dass geht in dem ioBroker Adapter ist dass möglich.

LG

sheyden, 2025-04-04:
@Maggi0r Can you share the MQTT topic/attribute names for that then easy enough to have a look at it.

Maggi0r, 2025-04-04:
Hello this is they are the Sensors in ioBroker Serialnumber is the Serialnumber from the Powerpacks. 
temperature (sensor packData.serialnumber.maxTemp) and battery SoC (sensor packData.serialnumber.socLevel), as well as voltage (sensor packData.serialnumber.totalVol).

I hope this is ok.

Greets Maggi

FireSon, 2025-04-08:
Since there is already another issue for this equest I will close this one. I will look into it once the smart matching mode is stable

closed 2025-04-08 by FireSon