Zendure/Zendure-HA #40: DeviceClass, Unit and Measurement for SolarInput

Bastix87, 2025-04-04:
Hello,

at first thank you for this great little integration. 
Maybe you can set the device_class, state_class and unit for the solarinput sensors ?

I´ve done it within my customization.yml but maybe you can add it to the integration ?

My example:
sensor.hub_2000_solarpower1:
  state_class: measurement
  unit_of_measurement: W
  device_class: power
  
sensor.hub_2000_solarpower2:
  state_class: measurement
  unit_of_measurement: W
  device_class: power

sheyden, 2025-04-04:
Can you share a list of entities you see exposed for the Hub2000 (based on sensor names)? They are currently not fully listed in the code and assigning the values can then easily be done from that listing.

FireSon, 2025-04-04:
Thanks for the input, I will look at it! Just finished some refactoring with the entities, so hopfully this wil help!

z-master42, 2025-04-17:
I think it would be nice if the sensors that only provide a numerical value between 0 and 1 or 2 had a more readable status.

FireSon, 2025-04-19:
@z-master42 when you know what these sensors are and what the values represent, please let me know and I will add them.

Most sensors have the correct setting for now, so I will close this issue. Please repopen it if there is something I have missed

closed 2025-04-19 by FireSon