Zendure/Zendure-HA #42: Smart matching does not work - always in high minus

Milvus11, 2025-04-04:
Hey FireSon, 
I am sorry to open another issue with it, I hope that someone tells me I do something wrong: Whenever I use Smart Matching, It goes way lower then -100 Watt. I use the tibber input for matching.
What can be the reason of that? I also tried a local P1 point, which goes into plus and minus and does not stop at 0, but behavior is the same. I attach the last 6 hours as a picture spike at 12.30 was the stove.

[image]

sheyden, 2025-04-04:
As someone who also went through quite a bit of the pain on this a few things to check:
- The meter you use to report as P1 data in Home assistant is it in W (so not kW!)?
- Do you have the Smart mode on, and then do not touch the feed-in or out mode in the actual device.

My PR for a slight improvement on docs is pending here: https://github.com/FireSon/Zendure-HA/blob/921b8f5197d2d4e979ff5bc960198a4302b2527b/README.md

But you can already use it to review and maybe figure out what might be going on.

FireSon, 2025-04-04:
I will, allmost finished my real job for today! But with a quick glance it looked pretty good :)

Maggi0r, 2025-04-04:
Hello, I have a question about how exactly the Power Manager works.
I only have one Hyper with two batteries.
Does it only feed in as much power as needed, like the CT mode with extra shelly? Or how exactly does it work? I'm currently controlling my feed-in using my smart meter and Node-Red, but it's not quite working yet.

Regards, Maggi

Milvus11, 2025-04-05:
Hey @sheyden, so I now let it run for a while, same problem, am at -130 W whole time (or, if sun is not enough, it lets everything in the grid, then minus 70/80 atm). Here my configurations, maybe you see something there?

1. Settings on the main page
[image]

2. Sensor acc. to your manual, which goes in positive and negative Watts
[image]

3. Sensor Settings of template
[image]

4. Sensor output
[image]

Milvus11, 2025-04-06:
maybe someone an idea?

closed 2025-04-08 by Milvus11