Zendure/Zendure-HA #43: Trigger the charging and discharging based on a electricity price

Andreasq22, 2025-04-04:
I want to charge the batteries if the price is low and discharge if the price is high.
I have also a percent value of the price calculated for the day (example: I want to charge if the percent of the max price is under 25% of the day and want to discharge if it ist over 75% of the price of the day.
I have all of this in home assistant, but how can I get the Hyper to charge/discharge?

jeff1975, 2025-04-05:
Ac laden wäre super wenn das funktioniert Hyper

sheyden, 2025-04-06:
@Andreasq22 You need to use the AC input and AC output mode to decide between those 2 modes in your automations and then you can use the sliders to decide how much power you need.

This is not the type of stuff an integration sorts out for you, but it is part of building automations: https://www.home-assistant.io/docs/automation/

Raudi1, 2025-04-06:
The Zendure App has a feature which charges/discharges based on the price. So it wouldn't be bad if we could control those settings from HA.
For my use case it would be enough to tell the Hyper via HA to start charging with full power(when the price is low as determined by some automation in HA or even EOS),
start discharging with full power(when the Zendure is full, the sun is shining and I plan to drive somewhere with my EV soon, I would rather have the energy in my car than donating it to the grid)
and switch back to matching the output with my Shelly pro 3 EM.

Andreasq22, 2025-04-06:
Yes, but:
Zendure can control the charging and discharging only by price based on one price. I want to have it to charge when under a lower price and discharge on a higher (second) price. This could be done in the app only in the "not CT"mode, but in the only price mode, in can not control the output based on Shelly. My dream is: control it by a factor (as i wrote), because one day a price is defined low as on other day... I don not want to define every day again, which price is low or hi.
I have all that infos in HA. I can also do automations. But I only see the control of charge or discharge (this I have not found) in this integration and a fix defined rate to do that. I need to have it based on then needed electricity

Raudi1, 2025-04-06:
I agree, the Zendure implementation is way too basic. But if you could set the price limit via HA, it would make things much easier.
Then the only issue is no surplus charging from AC, since Zendure only allows price, time or surplus charging separately.
The ultimate goal would be to control everything locally via HA and not to use any Zendure control at all. Then integrate everything with EVCC and EOS as well and we would have a completely automatic system which actually works and does what it should.

snipah, 2025-04-08:
For price-based charging and discarging I wrote automations in HA.

Below you see the automation for price based discharging, which relies on the tibber integration. You need to define a sensor for the tibber price level (VERY_CHEAP, CHEAP, NORMAL, EXPENSIVE, VERY_EXPENSIVE).

Basic rules:
- When price is "CHEAP", prioritize charging of battery
- When price is "NORMAL" use 80% solar & 20% from the wall
- When price is "HIGH" use solar/battery
- Wenn Battery is 100% and price is "CHEAP" only use Solar Yield

...see code for other logic.

Also switches for "Zero feed in" and "price based feed in" to be more flexble.

Maybe it is possible to use something like this in the Plugin...

```
alias: >-
  Zendure Outputlimit basierend auf Tibber Preisniveau, Nulleinspeisung &
  Prozentpuffer
description: >-
  Passt das Outputlimit basierend auf Leistungsdifferenz, Tibber Preisniveau,
  einem Prozent-Puffer an.
triggers:
  - entity_id:
      - sensor.shellypro3em_34987a4662f8_total_active_power
      - sensor.tibber_price_level
      - input_boolean.price_based_output_reduction
      - sensor.hyper_2000_electric_level
      - sensor.hyper_2000_solar_input_power
      - input_number.output_limit_buffer
    trigger: state
conditions:
  - condition: state
    entity_id: input_boolean.nulleinspeisung
    state: "on"
actions:
  - target:
      entity_id: number.hyper_2000_limit_output
    data:
      value: >
        {% set total_power =
        states('sensor.shellypro3em_34987a4662f8_total_active_power') | float(0)
        %} {% set home_power = states('sensor.hyper_2000_output_home_power') |
        float(0) %} {% set solar_yield =
        states('sensor.hyper_2000_solar_input_power') | float(0) %} {% set
        price_level = states('sensor.tibber_price_level') %} {% set
        battery_level = states('sensor.hyper_2000_electric_level') | int(0) %}
        {% set price_based =
        is_state('input_boolean.price_based_output_reduction', 'on') %} {% set
        buffer_percent = states('input_number.output_limit_buffer') | float(0)
        %} {% set min_output = 0 %} {% set base_output_limit = home_power +
        total_power %}

        {% macro apply_buffer(val) %}
          {{ val * (1 + buffer_percent / 100) }}
        {% endmacro %}

        {% if not price_based %}
          {{ apply_buffer(base_output_limit) | round(0, 'floor') }}
        {% elif battery_level == 100 %}
          {% if price_level in ['VERY_CHEAP', 'CHEAP'] %}
            {{ solar_yield | round(0, 'floor') }}
          {% else %}
            {% set multiplier = {
              'NORMAL': 0.80,
              'EXPENSIVE': 1.00,
              'VERY_EXPENSIVE': 1.00
            }.get(price_level, 1.00) %}
            {{ [(base_output_limit * multiplier) | round(0, 'floor'), min_output] | max }}
          {% endif %}
        {% else %}
          {% set multiplier = {
            'VERY_CHEAP': 0.00,
            'CHEAP': 0.00,
            'NORMAL': 0.80,
            'EXPENSIVE': 1.00,
            'VERY_EXPENSIVE': 1.00
          }.get(price_level, 1.00) %}
          {{ [(base_output_limit * multiplier) | round(0, 'floor'), min_output] | max }}
        {% endif %}
    action: number.set_value
mode: single
```

closed 2025-08-14 by FireSon