Zendure/Zendure-HA #44: de.json

Wunni73, 2025-04-04:
{
  "config": {
    "title": "Zendure Integration",
    "abort": {
      "already_configured": "Gerät ist bereits konfiguriert",
      "reconfigure_successful": "Neukonfiguration erfolgreich"
    },
    "error": {
      "cannot_connect": "Verbindung fehlgeschlagen",
      "invalid_auth": "Ungültige Authentifizierung",
      "unknown": "Unerwarteter Fehler"
    },
    "step": {
      "user": {
        "data": {
          "username": "Zendure Benutzername",
          "password": "Zendure Passwort",
          "broker": "Lokaler MQTT Broker",
          "p1meter": "P1 Sensor für Smart Matching",
          "phase1": "MMaximale Ausgabe für Phase 1",
          "phase2": "Maximale Ausgabe für Phase 2",
          "phase3": "Maximale Ausgabe für Phase 3"
        }
      },
      "reconfigure": {
        "data": {
          "username": "Zendure Benutzername",
          "password": "Zendure Passwort",
          "broker": "Lokaler MQTT Broker",
          "p1meter": "P1 Sensor für Smart Matching",
          "phase1": "Maximale Ausgabe für Phase 1",
          "phase2": "Maximale Ausgabe für Phase 2",
          "phase3": "Maximale Ausgabe für Phase 3"
        }
      }
    }
  },
  "options": {
    "step": {
      "init": {
        "data": {
          "scan_interval": "Scan Intervall in Sekunden"
        },
        "description": "Optionen anpassen",
        "title": "Zendure Integrationsoptionen"
      }
    }
  }
}

FireSon, 2025-04-04:
Nice vielen Dank! Ich habe angefangen die bezeignungen auch uebersetzbar zu machen. In die pre-29 version is diese de.json vervugbar, kannst versuchen die zu Uebersetzen?

Wunni73, 2025-04-05:
"entity": {
    "binary_sensor": {
      "master_switch": {
        "name": "Hauptschalter"
      },
      "wifi_state": {
        "name": "WLAN Status"
      }
    },
    "number": {
      "input_limit": {
        "name": "Eingangslimit"
      },
      "output_limit": {
        "name": "Ausgangslimit"
      },
      "manual_power": {
        "name": "Manuelle Leistung"
      },
      "min_soc": {
        "name": "SOC Minimum"
      },
      "soc_set": {
        "name": "SOC Maximum"
      }
    },
    "sensor": {
      "ambient_light_color": {
        "name": "Ambient Lichtfarbe"
      },
      "auto_model": {
        "name": "Batterie Programm",
        "state": {
          "0": "Kein",
          "6": "Batterieprioritätsmodus",
          "7": "Zeitliche Leistungsregelung",
          "8": "Smart Leistungsregelung",
          "9": "Smart CT Leistungsregelung",
          "10": "Strompreis"
        }
      },
      "pack_num": {
        "name": "Batterieanzahl"
      },
      "bat_volt": {
        "name": "Batteriespannung"
      },
      "hub_state": {
        "name": "Hub-Status"
      },
      "electric_level": {
        "name": "Batterie“
      },
      "charging_mode": {
        "name": "Lademodus",
        "state": {
          "0": "Nichts",
          "1": "mode 1",
          "2": "mode 2"
        }
      },
      "charging_type": {
        "name": "Ladetyp",
        "state": {
          "0": "Type 0",
          "1": "Type 1",
          "2": "Type 2"
        }
      },
      "charging_time": {
        "name": "Ladezeit"
      },
      "pack_state": {
        "name": "Batterie Status“,
        "state": {
          "0": "Ruhezustand",
          "1": "Laden",
          "2": "Entladen"
        }
      },
      "pack_input_power": {
        "name": "Batterieausgangsleistung"
      },
      "output_pack_power": {
        "name": "Batterieeingangsleistung"
      },
      "hyper_tmp": {
        "name": "Batterietemperatur"
      },
      "strength": {
        "name": "WLAN Signalstärke"
      }
    },
    "select": {
      "operation": {
        "name": "Betriebsmodus",
        "state": {
          "off": „aus“,
          "manual": "Manuelle Leistungsregelung",
          "smart": "Smarte Leistungsregelung"
        }
      },
      "ac_mode": {
        "name": "AC Betriebsmodus",
        "state": {
          "input": "AC Eingangsmodus",
          "output": "AC Ausgangsmodus"
        }
      }
    },
    "switch": {
      "lamp_switch": {
        "name": "Led"
      }
    }
  }
}

FireSon, 2025-04-06:
Added

closed 2025-04-06 by FireSon

Wunni73, 2025-04-06:
{
  "config": {
    "abort": {
      "already_configured": "Gerät ist bereits konfiguriert",
      "reconfigure_successful": "Neukonfiguration erfolgreich"
    },
    "error": {
      "cannot_connect": "Verbindung fehlgeschlagen",
      "invalid_auth": "Ungültige Authentifizierung",
      "unknown": "Unerwarteter Fehler"
    },
    "step": {
      "user": {
        "data": {
          "username": "Zendure Benutzername",
          "password": "Zendure Passwort",
          "broker": "Lokaler MQTT Broker",
          "p1meter": "P1 Sensor für Smart Matching",
          "phase1": "Maximale Ausgabe für Phase 1",
          "phase2": "Maximale Ausgabe für Phase 2",
          "phase3": "Maximale Ausgabe für Phase 3"
        }
      },
      "reconfigure": {
        "data": {
          "username": "Zendure Benutzername",
          "password": "Zendure Passwort",
          "broker": "Lokaler MQTT Broker",
          "p1meter": "P1 Sensor für Smart Matching",
          "phase1": "Maximale Ausgabe für Phase 1",
          "phase2": "Maximale Ausgabe für Phase 2",
          "phase3": "Maximale Ausgabe für Phase 3"
        }
      }
    }
  },
  "options": {
    "step": {
      "init": {
        "data": {
          "scan_interval": "Scan Intervall in Sekunden"
        },
        "description": "Optionen anpassen",
        "title": "Zendure Integrationsoptionen"
      }
    }
  },
  "entity": {
    "binary_sensor": {
      "master_switch": {
        "name": "Hauptschalter"
      },
      "wifi_state": {
        "name": "WLAN Status"
      }
    },
    "number": {
      "input_limit": {
        "name": "Eingangslimit"
      },
      "output_limit": {
        "name": "Ausgangslimit"
      },
      "manual_power": {
        "name": "Manuelle Leistung"
      },
      "min_soc": {
        "name": "SOC Minimum"
      },
      "soc_set": {
        "name": "SOC Maximum"
      }
    },
    "sensor": {
      "ambient_light_color": {
        "name": "Ambient Lichtfarbe"
      },
      "auto_model": {
        "name": "Batterie Programm",
        "state": {
          "0": "Kein",
          "6": "Batterieprioritätsmodus",
          "7": "Zeitliche Leistungsregelung",
          "8": "Smart Leistungsregelung",
          "9": "Smart CT Leistungsregelung",
          "10": "Strompreis"
        }
      },
      "pack_num": {
        "name": "Batterieanzahl"
      },
      "bat_volt": {
        "name": "Batteriespannung"
      },
      "hub_state": {
        "name": "Hub-Status"
      },
      "electric_level": {
        "name": "Batterie"
      },
      "charging_mode": {
        "name": "Lademodus",
        "state": {
          "0": "Nichts",
          "1": "Mode 1",
          "2": "Mode 2"
        }
      },
      "charging_type": {
        "name": "Ladetyp",
        "state": {
          "0": "Type 0",
          "1": "Type 1",
          "2": "Type 2"
        }
      },
      "charging_time": {
        "name": "Ladezeit"
      },
      "pack_state": {
        "name": "Batterie Status",
        "state": {
          "0": "Ruhezustand",
          "1": "Laden",
          "2": "Entladen"
        }
      },
      "pack_input_power": {
        "name": "Batterieausgangsleistung"
      },
      "output_pack_power": {
        "name": "Batterieeingangsleistung"
      },
      "hyper_tmp": {
        "name": "Batterietemperatur"
      },
      "strength": {
        "name": "WLAN Signalstärke"
      }
    },
    "select": {
      "operation": {
        "name": "Betriebsmodus",
        "state": {
          "off": "Aus",
          "manual": "Manuelle Leistungsregelung",
          "smart": "Smarte Leistungsregelung"
        }
      },
      "ac_mode": {
        "name": "AC Betriebsmodus",
        "state": {
          "input": "AC Eingangsmodus",
          "output": "AC Ausgangsmodus"
        }
      }
    },
    "switch": {
      "lamp_switch": {
        "name": "Led"
      }
    }
  }
}

Wunni73, 2025-04-06:
sorry ich habe noch 3 Fehler gefunden :-(