Zendure/Zendure-HA #45: Hub 1200 läßt sich nicht einbinden

jeff1975, 2025-04-05:
Danke für diese super Indegration leider läst sich mit Relaise 29 der Hub 1200 nicht einbinden Hyper wird gefunden Hub 1200 nicht

SleXx88, 2025-04-05:
I have the same problem with the SF800 on the 29th version, it is not recognized.

[image]
[image]

FireSon, 2025-04-05:
Thanks, I saw it. It will be in the next pre-release. Please note that these versions are not fully tested and thinks like this can happen ;-)

SleXx88, 2025-04-05:
I am very grateful that you provide this integration. Many thanks 😃👌🏻

Eagel1959, 2025-04-06:
Erstmal danke
Gerade installiert und funktioniert auch, habe ein Hub2000 und ein Hub1000 , wobei mir aber nur der 2000er angezeigt wird.
Eine Idee ?

FireSon, 2025-04-06:
@Eagel1959 I do not have these devices so that makes it difficult to test. Can you test the prerelease 1.0.29 and see it you can find it there?

Eagel1959, 2025-04-06:
Bin absoluter Anfänger. Werde es aber versuchen

SleXx88, 2025-04-06:
@FireSon ich habe den HUB1200 und dem SF800. Ich könnte den HUB1200 in den nächsten Tagen testen.

kde1806, 2025-04-07:
@FireSon Hallo ich bin happy mit der Integration aber der Hub1200 wird noch nicht in der 1.0.29-pre-release im Zendure Account erkannt. Der Hyper2000 funktioniert.

Eagel1959, 2025-04-07:
OK ,,danke @

seb8888, 2025-04-07:
For the hub1200 the name is not "Hub 1200" but "SolarFlow2.0"
[image]

jeff1975, 2025-04-12:
DANKE für Deine Arbeit !!!!!
Hub 1200 funktioniert perfekt mit Version 1.29

closed 2025-04-16 by FireSon