Zendure/Zendure-HA #52: Option for Grid-Based Charging Prioritization in Smart Matching System

Popoboxxo, 2025-04-06:
Zendure-HA currently optimizes self-consumption by discharging the battery within the Smart Matching System. However, during periods of very low electricity tariffs, it might be beneficial to have the option to avoid battery discharge and instead utilize only excess grid power for charging.

Proposal:

It would be helpful if an option could be added to adjust the charging priority within the Smart Matching System. This could be achieved through an additional parameter or switch in the manager interface, for example.

Possible Benefits:

Improved Cost Efficiency: Maximize savings during periods of very low electricity tariffs.
Reduced Efficiency Losses: Avoid unnecessary battery discharge.
Increased Flexibility: Users could better adapt the system to their individual needs and current electricity prices.

Possible Implementation Ideas:

A simple switch in the user interface, e.g., "Prioritize Grid Charging. - with the idea to implement an Automation with the parameter to set a price threshold above which battery discharge is allowed. Alternatively, perhaps an automation could enable charging during low solar days at cheap electricity prices and discharging at expensive electricity prices, if the price difference is significant enough to offset the losses

Perhaps it's possible to utilize and zero the P1 sensor input to avoid exiting smart mode and quickly re-implement changes in excess powe

BR

Daniel

Popoboxxo, 2025-04-23:
Hey @FireSon !

"Currently, I'm trying to influence the behavior of the Smart Mode using an influenceable measurement sensor. For example, I'm simulating that power is only being fed in, so that I force the system to only charge excess power instead of also discharging.

However, I'm still stuck on the fact that the Smart Mode apparently can't handle it when a 0 Watt value is transmitted at an unfortunate moment. It then continues charging instead of doing 'nothing'.

Can you give me a tip on when the Smart Mode starts to do 'nothing' - that is, puts the battery in standby? I'm currently trying it with an intermediate state of the P1 sensor, which then oscillates between 15 and -5 watts every second. That works semi-successfully."

My Current experiments:

`- sensor:
    - name: DynamischesNetzlastVerhalten
      unit_of_measurement: `"W"`
      unique_id: 9eae0559-60c4-48b8-8765-cb0686bfe251
      state: >-
        {% set modus = states('input_select.netzmessmodus') %}
        {% if modus == 'Normal' %}
          {{ states('sensor.aktuelle_netzlast') }}
        {% elif modus == 'Negativ' %}
          {% set pv_einspeisung = states('sensor.pv_einspeisung_ausgelesen_local') | float %}
          {% if pv_einspeisung == 0 %}
            {{ 10 }}
          {% else %}
            {{ pv_einspeisung }}
          {% endif %}
        {% elif modus == 'Positiv' %}
          {% set verbrauch = states('sensor.total_power_nur_verbrauch_local') | float %}
          {% if verbrauch == 0 %}
            {{ -10 }}
          {% else %}
            {{ verbrauch }}
          {% endif %}
        {% elif modus == 'Statisch Negativ' %}
          {{ states('sensor.pv_einspeisung_ausgelesen_local') | float - 1200 }}
        {% elif modus == 'Statisch Positiv' %}
          {{ states('sensor.total_power_nur_verbrauch_local') | float + 800 }}
        {% elif modus == 'Aus' %}
          {{ states('sensor.schwingender_wert_5_sekunden') }}
        {% else %}
          {{ 0 }}
        {% endif %}`

and 

`- trigger:
    - trigger: time_pattern
      seconds: /2
  sensor:
    - name: "Schwingender Wert (5 Sekunden)"
      unique_id: schwingender_wert_5s_modern
      state: >
        {% set sekunde = now().second %}
        {% set offset = ((sekunde / 5) % 2) | round(0) * 10 - 5 %}
        {{ offset }}`

Not elegant, but works in 70% of the cases. :D

An "Only Changing Smart Modus" and "Only Discharging Smart Modus" would be awesome

Popoboxxo, 2025-05-29:
I'll force the "last setpoint" to 0, to disable the discharging in Smart-Mode. But i thing, that the Function mabe stuck in some situations in an charging/discharging phase.

Popoboxxo, 2025-06-15:
Hey!

I've been tinkering a bit these past few days and finally got EOS (https://github.com/Akkudoktor-EOS/EOS) up and running. In the evenings, when the optimization system suggests preventing battery discharge, I manage perfectly fine with the manual mode. However, if a "charge yes - discharge no" situation is recommended during midday hours, I'm currently only able to manage it poorly with a workaround.

An Only Load Smart Mode would therefore be amazing.

Here's an example from the optimization:

[image]

I'm currently using:

https://github.com/Duetting/ha_eos_addon
https://github.com/ohAnd/EOS_connect

harrymayr, 2025-06-15:
> Only Load Smart Mode

Next release 😉😀

Popoboxxo, 2025-06-15:
Ohhh boy!!! Im thrilled! Hype Train activated 🚄

closed 2025-08-14 by FireSon