Zendure/Zendure-HA #53: Request for explanation of local mode (pre1.0.29)

sankukace, 2025-04-07:
I just installed pre1.0.29 on a separate VM to avoid messing things up.

How does it work locally? I only noticed the translation — everything else seems the same. How do you connect to the local setup?

It simulates a Shelly, yes, but if there's no internet, will it still work? Just trying to understand.

Thanks again.

FireSon, 2025-04-08:
Hi @sankukace some time ago, Zendure released a firmware in which the Shelly3EM was able to communicate locally over wifi, with the Hyper2000. Exactly the same thing the integration is doing now as well. So the integration sends exactly the same commands as the Shelly does, to the Hyper2000. 
This is what in the Zendure App is called CT mode. Normally you have to select a Shelly, but the integration takes care of that. It uses the "Follow the Surplus" option to do both charging and discharging.
But you are right, you should not notice any difference, since the integration takes care of it :-)

sankukace, 2025-04-09:
Thanks, I thought it was the local MQTT version that was coming :)
I believe it's planned.

Thanks again!

closed 2025-04-09 by sankukace