Zendure/Zendure-HA #56: Integrate Smart-Meter / Smart Plugs into HA

lpetflo, 2025-04-08:
First of all thank you very much for your efforts regarding the HA integration of the Zendure system! Your work did really help me out.

One minor problem (or better room for improvement) would be the integration of Zendure Smart Plugs from the App to Home Assistant.
Besides the Shelly Smart Plugs, some way cheaper Tuya based Plugs (in my case Eightree Plugs at ~8€ per Plug) can be integrated as mockup Zendure Plugs. The downside is, that they cannot be integrated into the Tuya system at the same time. Therefore for monitoring the energy consumption of a device, while using the energy consumption to trigger the plug based automation mode in the zendure app, would require a seperate additional smart plug for HA. Two plugs per device seems to be a bit over the top so I want to use the Zendure one only.

Do you by any means see a way to read the data from the plugs and provide them to the HA-integration?
I will take a look at the code as well, but as I'm not to deeps into HACS I'n not sure if and how I could provide guidance.

Best regards and keep up the amazing work :)

[image]

FireSon, 2025-04-08:
@lpetflo  Nice idea, I will look into it, once the smart matching mode works stable.

Milvus11, 2025-04-08:
Hey @FireSon, mind if I ask, what is not stable at smart matching atm?

lpetflo, 2025-04-09:
I just tried finding a corresponding message similar to your explanation in #6, but when changing the smart plan (based on the smart plugs) I don't see a corresponding message in the log. Only the basic save command:

`2025-04-09 20:57:56.867 INFO (paho-mqtt-client-xxx:xxx-xxx-xxx) [custom_components.zendure_ha.zendurermanager] Unknown topic: iot/gDa3tb/xxx/function/invoke => {'arguments': [{'autoModelProgram': 1, 'autoModelValue': {'upTime': 201, 'chargingType': 1, 'pullTime': 200, 'price': 33, 'chargingPower': 30, 'prices': [33, 32, 32, 32, 32, 34, 39, 42, 36, 33, 31, 28, 22, 22, 22, 25, 28, 31, 33, 36, 37, 34, 33, 32], 'outPower': 90}, 'msgType': 1, 'autoModel': 8}], 'deviceKey': 'xxx', 'function': 'deviceAutomation', 'messageId': 51063374, 'timestamp': 1744225076}`

When switching the plugs on/off no separate log message is send as well :(
Does this mean, that integrating the plugs won't work as easily or did I just not fully understand how this works? However the plug device itself isn't integrated into the Zendure-Plugin, so it might not be listening to messages from this device yet?

De-Elsaesser, 2025-05-16:
@FireSon  By the way, Zendure Satellite plugs Eu can be controlled via local Tuya. I could connect them like a Tuya device via the smart+ app from Ledvance and then control them via local Tuya integration. In this way, special Zendure plugs, connected via Zendure cloud are not needed.

De-Elsaesser, 2025-08-06:
@Gyvate108 Hi, I don't remeber how I did the integration from Zendure plug into local Tuya, but the raison was to act locally, without Zendure Cloud. You are right, in my configuration, the plugs can no longer be used in the app. I don't need it, I work directly with the HA app over VPN.

lpetflo, 2025-08-08:
@FireSon Any possibility that with the new official(?) Zendure collaboration the Plugs might be added to the Api and thus to this ZendureHA integration?
I currently even use the Zendure stacked with "real" smartplugs (directly connected to HA) to get the energy consumption info of my devices in HA.

MrFuellort, 2025-08-10:
@FireSon,

ich möchte mich zunächst für Ihre großartige Arbeit an der Zendure-Integration bedanken. 
Sie ist wirklich sehr hilfreich.

Ich habe eine Frage zur Integration. Ist geplant, den Smart Meter Zendure 3CT in naher Zukunft zu integrieren? Ich verwende diesen derzeit, aber er wird noch nicht unterstützt, daher kann ich keinen Autarkiegrad ermitteln und brauchbare Langzeit Werte sammeln.

Mir fehlen dafür die entsprechenden Sensoren für den Hausverbrauch, den Netzbezug und die Netzeinspeisung.

Falls es bereits eine interne Beta gibt, würde ich mich gerne als Tester anbieten. 

Vielen Dank vorab. 

——————————————————————————
First, I'd like to thank you for your great work on the Zendure integration.

It's really very helpful.

I have a question about the integration. Are there any plans to integrate the Zendure 3CT smart meter in the near future? I'm currently using it, but it's not yet supported, so I can't determine my level of self-sufficiency or collect useful long-term values.

I don't have the appropriate sensors for household consumption, grid consumption, and grid feed-in.

If there's already an internal beta, I'd be happy to volunteer to test it.

Thank you in advance.

Kieft-C, 2025-12-28:
@MrFuellort  : siehe https://github.com/Zendure/Zendure-HA/issues/803

Gyvate108, 2025-12-29:
@FireSon and others knowledgeable

> **I will ask Zendure if it is possible** 😄

That was 08-Aug-2025 - Any update here ?

When they manage to get their plugs into their app, it should also be possible to send this info to the HA integration.
At IFA 2025 in Berlin they confirmed it being possible...
Possible/feasible is one thing - doing it another

janssensjeroen, 2026-09-11:
I agree, that would be a nice enhancement

scubafly, 2026-11-12:
mqtt connection is non existing, can not read them / control them on ip.
So plugs are kinda useless to me if I want them in home assistant.