Zendure/Zendure-HA #59: Has anyone tried this integration with the SolarFlow 2400AC?

HomeAssistantPim, 2025-04-08:
Has anyone tried this integration with the SolarFlow 2400AC?
Or else, would anyone know if with some changes I could make it work (and contribute here)?

FireSon, 2025-04-08:
We have planned it, as soon as we have one to test. Probably in the next week!

FireSon, 2025-04-11:
Just tried the first steps! The device gets recognized and I added the exact same sensors and functionality of the Hyper2000. I understood that the device does not support device sharing, so it might be difficult to debug!

This will be in the version 29, which I will release soon!

asatron1, 2025-04-19:
Hi, wie ist denn der Stand mit der SolarFlow 2400AC? 
Habe das System noch nicht und bin am überlegen.
Liebe Grüße

FireSon, 2025-04-23:
@asatron1 we just tested it and added the SF2400 now 😄

closed 2025-04-25 by FireSon