Zendure/Zendure-HA #60: Outputlimit cannot be Set on Solar 800 Flow

thhoppst, 2025-04-09:
When I set the output limit on the slider nothing happens.

FireSon, 2025-04-10:
@thhoppst since I do not have one, it is hard to test. I have found the problem, so hopefully it will be fixed in the next release.

thhoppst, 2025-04-10:
I will test it for you :)

thhoppst, 2025-04-11:
Sorry. I clone the project from github and i install it into my HomeA. When i change the  outputlimit slider nothing will send to the solar flow 800,

[image]

FireSon, 2025-04-11:
@thhoppst can you install the latest version (29) and test it again?

thhoppst, 2025-04-11:
Ok hub 2000 is ok. I can only test the solar flow 800 tomorrow when solar energy is available again.

thhoppst, 2025-04-12:
I get the following error in Home A:

`2025-04-12 07:48:32.105 ERROR (paho-mqtt-client-30789:ac7e53da-aa29-40a1-b792-c0da2d40a0e6) [custom_components.zendure_ha.zendurermanager] 'NoneType' object has no attribute 'default_language_platform_translations'

2025-04-12 07:48:32.106 ERROR (paho-mqtt-client-30789:ac7e53da-aa29-40a1-b792-c0da2d40a0e6) [custom_components.zendure_ha.zendurermanager] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zendurermanager.py", line 159, in on_message
    if device.updateProperty(key, value):
       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/devices/solarflow800.py", line 75, in updateProperty
    if not super().updateProperty(key, value):
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty
    if sensor.state != value:
       ^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 501, in unit_of_measurement
    := self.platform.default_language_platform_translations.get(translation_key)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'default_language_platform_translations'`

closed 2025-04-14 by FireSon