Zendure/Zendure-HA #61: no sensors?

surfer1264, 2025-04-09:
Hey 
thanks for the integration
installation went smooth
after installation and restart of HA 1 device and 2 Entities (I own a Hub1200)
Is this correct?
I expected a lot of entities and switches?

Need to say,that I am using the zmaster42  approach for getting data (using MQTT, with defined entities in mqtt.yaml, and using appkey and appsecret provided by Zendure)

Here is the picture ..what I see after installation
[image]

Is this a problem when both "integrations are installed in parallel?
Do I need to remove the old one?

Thank you

FireSon, 2025-04-10:
@surfer1264 difficult to explain without debug logging. You can use it in 'parallel' but than you must probably have multiple shared accounts. An account can only be used by one connection. Maybe you can test it with your main account first?

surfer1264, 2025-04-10:
As recommended  I am using a separate Account for the Access
See log
Login successful
Why mentioned hyper?
I dont have an hyper, but SF1200

My System is Registered in zendure Region Global.
Could this be the Problem ?
That your Integration is trying to fetch data from the EU Region ?

2025-04-10 22:26:09.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh hypers
2025-04-10 22:26:09.014 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching homeassistant (Zendure Integration - surfer1264@bkw.————.de) data in 0.001 seconds (success: True)
2025-04-10 22:27:39.013 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh hypers
2025-04-10 22:27:39.014 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching homeassistant (Zendure Integration - surfer1264@bkw.————.de) data in 0.001 seconds (success: True)

surfer1264, 2025-04-10:
Next try with my original Master Account following log
Many sensors and informations in Log but no Sensors in HA See pictures 
(Replaced serialnumbers and device ids with x) 

[image]

2025-04-10 23:08:59.605 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 1 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 23:08:59.605 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 2 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 23:08:59.605 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 3 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 23:08:59.606 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-04-10 23:08:59.611 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-04-10 23:08:59.612 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-04-10 23:08:59.908 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/v2 => Mqtt: mq.zen-iot.com
2025-04-10 23:08:59.909 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-04-10 23:09:00.412 INFO (MainThread) [custom_components.zendure_ha.api] Getting device details for [51237] ...
2025-04-10 23:09:00.997 INFO (MainThread) [custom_components.zendure_ha.api] Got data for [51237] 4...
2025-04-10 23:09:00.997 INFO (MainThread) [custom_components.zendure_ha.api] Adding device: xxxxxxxx SolarFlow2.0
2025-04-10 23:09:00.997 INFO (MainThread) [custom_components.zendure_ha.api] Device SolarFlow2.0 is not supported!
2025-04-10 23:09:00.998 INFO (MainThread) [custom_components.zendure_ha.api] Data: {'isShare': False, 'productKey': '73bkTV', 'productModelId': 28, 'productName': 'SolarFlow2.0', 'deviceName': 'SolarHerwegh', 'deviceId': 51237, 'deviceKey': 'KH6E9v9x', 'snNumber': 'xxxxxxxxxxxxxxxx', 'onlineFlag': '00601', 'standard': '', 'consumerId': 22619, 'shortCode': 'P1', 'productType': 8, 'priceType': '', 'id': 10275, 'masterSwitch': True, 'restState': 0, 'restTime': '', 'hubState': 1, 'wifiState': True, 'wifiName': 'PrivKabelNew5G', 'mac': '94:c9:60:4d:cc:74', 'ip': '192.168.178.95', 'buzzerSwitch': True, 'autoHeat': 0, 'loraInvState': 0, 'loraAddr': '', 'ambientSwitch': False, 'ambientLightMode': 0, 'ambientLightColor': 0, 'ambientLightNess': 0, 'solarInputPower': 0, 'pvPower1': 0, 'pvPower2': 0, 'pvPower3': 0, 'pvPower4': 0, 'packInputPower': 180, 'outputPackPower': 0, 'outputHomePower': 172, 'outputInversePower': 0, 'bindSn': '', 'invOutputPower': 0, 'outputLimit': 60, 'inputLimit': 0, 'remainOutTime': 793, 'remainInputTime': 59940, 'socSet': 1000, 'minSoc': 170, 'packState': 2, 'packNum': 2, 'electricLevel': 70, 'isDeleted': 0, 'createTime': '2024-03-29T12:09:05Z', 'updateTime': '2025-04-10T21:08:43Z', 'inverseMaxPower': 500, 'pvBrand': 0, 'brandId': 1, 'cycle': 0, 'blueOta': 1, 'solarInputPowerCycle': 0, 'solarInputPower1Cycle': 0, 'solarInputPower2Cycle': 0, 'packInputPowerCycle': 0, 'outputPackPowerCycle': 0, 'outputInverseCycle': 0, 'outputHomePowerCycle': 0, 'gridInputPowerCycle': 0, 'gridInputPower': 0, 'smartMode': 0, 'smartPower': 0, 'pass': 0, 'lowTemperature': 0, 'heatState': 0, 'modelName': 'Hoymiles-HM-800', 'passMode': 0, 'lowTempNotice': 2, 'autoRecover': 1, 'acMode': 2, 'acDelay': 0, 'acSwitch': 0, 'dcSwitch': 0, 'acOutputPower': 0, 'dcOutputPower': 0, 'acOutputPowerCycle': 0, 'dcOutputPowerCycle': 0, 'socStatus': 0, 'gridStandard': '', 'phaseCheck': None, 'priority': 0, 'gridReverse': 0, 'reverseState': 0, 'energyPower': 0, 'lampSwitch': 0, 'clusterSw': 0, 'hyperTmp': 0, 'plugState': 0, 'gridOffMode': 0, 'localState': 0, 'panelType': 0, 'panelPower': 0, 'panelNums': 0, 'fanSwitch': 0, 'fanSpeed': 0, 'outputPower': 0, 'gridState': 0, 'chargePower': 0, 'faultNum': 0, 'priceUpdateDate': '', 'gridOffPower': 0, 'socLimit': 0, 'signUrl': '', 'signCode': '', 'aiSet': '', 'aiState': 0, 'packDataList': [{'createTime': '2024-09-30T03:18:19Z', 'updateUser': 0, 'updateTime': '2025-04-10T21:09:00.865Z', 'isDeleted': 0, 'id': 685105, 'deviceId': 51237, 'consumerId': 22619, 'socLevel': 69, 'state': 2, 'maxTemp': 293.1, 'batteryCode': ' xxxxxxxxxxxxxxxxx', 'softVersion': 4113, 'version': '1.0.17', 'sn': 'xxxxxxxxxxxxxxxx', 'heatState': 0, 'packType': 2, 'packName': '', 'packModel': ''}, {'createTime': '2025-03-18T06:58:16Z', 'updateUser': 0, 'updateTime': '2025-04-10T21:09:00.865Z', 'isDeleted': 0, 'id': 921631, 'deviceId': 51237, 'consumerId': 22619, 'socLevel': 71, 'state': 2, 'maxTemp': 290.1, 'batteryCode': 'xxxxxxxxxxxxxxxx', 'softVersion': 8223, 'version': '2.0.31', 'sn': 'xxxxxxxxxxxxxxxxx', 'heatState': 0, 'packType': 3, 'packName': '', 'packModel': ''}], 'todayEnergy': 3367.13, 'temperatureUnit': 0, 'faultFlag': False, 'bateUser': False, 'deviceId2': 51237, 'autoModel': 8, 'solarPower1': 0, 'solarPower2': 0, 'bindProduct': None, 'bindName': '', 'bindDeviceKey': '', 'bindDeviceState': '', 'bindId': None, 'aceSolarPower': 0, 'clusterId': None, 'nodeStatus': None, 'clusterStatus': None, 'hyperNum': 0, 'hemsId': None, 'hemsShareFlag': False, 'remindTime': '', 'networkType': 0, 'energyDate': '2025-04-10', 'otaBatch': False, 'hemsName': ''}

surfer1264, 2025-04-11:
Maybe this Problem ?
2025-04-10 23:09:00.997 INFO (MainThread) [custom_components.zendure_ha.api] Adding device: xxxxxxxx **SolarFlow2.0**
2025-04-10 23:09:00.997 INFO (MainThread) [custom_components.zendure_ha.api] **Device SolarFlow2.0 is not supported!**

FireSon, 2025-04-11:
Yes that is probably the reason, I will add it to the new release!

surfer1264, 2025-04-11:
Seems to work

[image]

closed 2025-04-11 by FireSon