Zendure/Zendure-HA #62: Add Energy Dashboard-compatible sensors

deceju, 2025-04-10:
Hello,
Thanks for your work on the Zendure integration.

Currently, the integration exposes basic power sensors, but it’s missing key values needed for full compatibility with Home Assistant’s Energy Dashboard.

Other inverter integrations (like SolarEdge, Enphase, etc.) provide native sensors that are directly usable in the Energy dashboard, without requiring users to manually create template or integration sensors.

It would be great if this integration could expose values such as:
	•	Solar power used directly by the home (not stored)
	•	Solar power sent to battery
	•	Battery charge from solar
	•	Battery charge from grid
	•	(and their energy equivalents in kWh, if possible)

These are just suggestions based on common usage patterns — I’m not sure which specific values need to be exposed for the integration to be fully plug-and-play with the Energy Dashboard, but the goal is to avoid having to rebuild all energy flows manually with templates.

Thanks in advance !

SleXx88, 2025-04-10:
Genau an diesem Punkt arbeite ich gerade und werde bei Fertigstellung einen Pull-Request eröffnen.

FireSon, 2025-04-18:
@deceju In version 36 I have added two sensors aggrChargeDaykWh and aggrDischargeDaykWh can you check if that is what you need?

GeneralRurra, 2025-04-19:
I think there's something wrong - that's the current status

[image]

[image]

GeneralRurra, 2025-04-19:
Update 2 

[image]

[image]

GeneralRurra, 2025-04-22:
It would be nice if you could add a daily yield sensor, then I think everything would be complete as far as the energy dashboard is concerned

FireSon, 2025-04-25:
@GeneralRurra the two aggr sensor should reset each day. Can you check if that is working properly?

GeneralRurra, 2025-04-25:
[image]

No, they do not

GeneralRurra, 2025-04-25:
[image]

No, they do not 
@FireSon

GeneralRurra, 2025-04-25:
If you want to make something to test for the Hyper, feel free to write to me, I love testing for my life

GeneralRurra, 2025-04-26:
Today one of the sensors reset itself at 0 o'clock 

[image]

harrymayr, 2025-04-30:
The latest commit (as of 29.04.25) will fullfil the defined needs. There is not yet a release, but with 1.0.41 it will work. It could take then 1-2 days until the Energy dashboard shows daily values.

harrymayr, 2025-05-01:
The sensor itself work now as expected, unfourtunatly they get a little bit crazy in the Energy Dashboard. Not sure, if the rootcause are the frequent restarts of HA at the moment.

GeneralRurra, 2025-05-01:
How do you mean play crazy?

harrymayr, 2025-05-01:
Shows 27,75kWh for aggrDischargeDaykWh but true is 2.6kWh. And for aggrChargeDaykWh it shows -59,87kWh instead of -3.66kWh. But if you click on the sensor in the dashboard, so that it show the day trend, it looks OK. (can't paste an image to show)

GeneralRurra, 2025-05-01:
Maybe at 0 o'clock when it resets it will be better

twoenter, 2025-05-02:
[image]
Discharges does reset at nidnight, charge doesn't. I'm on v40

harrymayr, 2025-05-04:
But on the other hand, I need to raise the question, if a sensor which will be reset every day is the right choise for the Energie-Dashboard. Every other sensor you add there is a sensor from a electricity or gas meter which will be reset only, if you need to exchange the meter. So perhaps adding a 2nd sensor-set which will not reseted, or change the exsiting senors to not reset. The Energy-Dashboard give you the daily value and if you want you can add a consumption meter in HA.

GeneralRurra, 2025-05-04:
The Home Assistant Dashboard doesn't really like reset sensors either, it's just a fine line, it can happen that it suddenly deducts 8kwh in the Energy Dashboard the next day because the sensor has reset itself to 0 
Suggestion 
2 kWh meter total_increasing without resetting 
2 kWh meter as they are already present 

And please add a meter for the solar input thanks

FireSon, 2025-05-11:
Fixed in 41-pre7

closed 2025-05-11 by FireSon