Zendure/Zendure-HA #63: Problem to configure with sharing account

florianfish, 2025-04-10:
Hello,
I'm having trouble configuring the Zendure integration on Home Assistant using the shared account. I've successfully shared the system to my secondary account, and I've validated the sharing from there. However, when I configure it from Home Assistant using the secondary account, I don't see any entities configured.

florianfish, 2025-04-10:
Hi @FireSon everithing is ok when i connect with my main account.
There are the logs : 

2025-04-10 16:31:40.967 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration zendure_ha which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-10 16:31:40.968 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-10 16:33:11.307 DEBUG (MainThread) [custom_components.zendure_ha.config_flow] Check API connection
2025-04-10 16:33:11.307 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-04-10 16:33:12.298 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
2025-04-10 16:33:12.299 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 1 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 16:33:12.299 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 2 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 16:33:12.299 INFO (MainThread) [custom_components.zendure_ha.zendurephase] Create phase 3 with 2400 watt output and 800 watt input 800 watt output
2025-04-10 16:33:12.299 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-04-10 16:33:12.300 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-04-10 16:33:12.300 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-04-10 16:33:13.053 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
2025-04-10 16:33:13.053 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-04-10 16:33:13.139 INFO (MainThread) [custom_components.zendure_ha.api] Create mqtt client!! 75701:02900acd-a3db-42ff-8728-a78575757929
2025-04-10 16:33:13.234 INFO (MainThread) [custom_components.zendure_ha.api] Client has been connected
2025-04-10 16:33:13.235 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found: 0 hypers
2025-04-10 16:33:13.235 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding sensors homeassistant (Zendure Integration - XXXXXX@gmail.com)
2025-04-10 16:33:13.237 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh hypers
2025-04-10 16:33:13.237 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching homeassistant (Zendure Integration - XXXXX@gmail.com) data in 0.000 seconds (success: True)

FireSon, 2025-04-10:
Did you share your devices correctly (share + receive)? From the logging:

> [custom_components.zendure_ha.zendurermanager] Found: 0 hypers

It seems there are no hypers available for the account...

florianfish, 2025-04-10:
@FireSon I have share the system with family member

SleXx88, 2025-04-10:
@florianfish The same problem for me.

SleXx88, 2025-04-10:
I use the Solarflow 800, how does sharing work in the Zendure app. The attachment is shared on my second email, but there is no entry under Account > Device sharing.If I use the second account, the integration only shows two entities.This is probably due to the permissions.

[image]
[image]

SleXx88, 2025-04-11:
@FireSon Yes, I noticed that too. But it is not bad. I can be logged in to Zendure Integration and the app with the main account without any problems. Both provide me with data without delay. Only when reloading the integration or restarting HA do you have to log in to the app. However, integration does not lose any connection. So, everything is great.💪🏼

FireSon, 2025-04-11:
Close this for now, since it is the best we can do!

closed 2025-04-11 by FireSon

florianfish, 2025-04-14:
I confirm that @SleXx88 said : if we log in in integration and then in the app, the two connection stay logged!