Zendure/Zendure-HA #66: Thank you

Smartzeug, 2025-04-10:
Hello FireSon,
I'm currently doing a review about this integration for my YouTube Channel Smartzeug (www.youtube.com/smartzeug).

I just wanted to say thank you 🥰
The video will be released on the 17th of April at 4pm Germany time. 

Is there a possibility where I can buy you a coffee or send you a tip via PayPal?

Keep up the good work 💪

Cheers,
Uğur

FireSon, 2025-04-10:
@Smartzeug Hi that is nice!! When will you record it? I am almost finished with a major update which will also change the configuration considerably. So it would be the bast to do it with the updated version!

Do you know what you will cover? Especially the second account with 'shared' devices is difficult, and the way (in the new version!) how to handle clusters would be good to explain. Also the way the power is distributed over multiple devices might be interesting. And the handling of the active_power sensor.

coffee is always good, I will look at it ;)

Smartzeug, 2025-04-10:
@FireSon actually I finished recording 😃
Because I have only 1 Hyper 2000 I could only explain theoretically the Power Manager. 
To be honest I don't think it is a big deal of having 2 accounts. I never looked into the Zendure App anyway after integration via MQTT or with your integration. 

Therefore I kept it basic 😉

Let me know your PayPal or Buy Me A Coffee so I can give you a tip please 🫶

closed 2025-04-11 by FireSon

FireSon, 2025-04-11:
@Smartzeug well I have added buy be a coffee to the home page of the integration. https://buymeacoffee.com/fireson