Zendure/Zendure-HA #67: Automatic calibration of battery included? QUESTION

Milvus11, 2025-04-11:
Hey, I experience an issue with the battery, there always is a slump from 22% to 10%. So i assume the calibration of the battery, which it supposedly does automatically, would be necessary. However, is the calibration working, while the HA integration is running? Or must it be implemented here? 
Can it be triggered somehow?

FireSon, 2025-04-11:
@Milvus11 I think it would be great to add it here if possible. I was under the impression that Zendure would take care of that. What would be the best way to add it?

Milvus11, 2025-04-12:
@FireSon, I checked through different websites now, if you ask 10 people, you get 15 answers... What seems to be the most mentioned point is to at least once a month charge the batteries to 100%. 
Some also mentioned that beforehand a discharge down to 0 should be done, but there are more comments against that to not crack up the battery. 

So conclusion I saw so far: every now and then, at least once a month, charging to 100%. Some People also mention, that from around 15% the line goes down non-linear. So seems I am not alone with my "problem" and it is a common thing.

So unless your developer-sources at Zendure don't say different things, I assume, that a Checkbox "Automatic battery calibration on/off" could be enough, which would be a date/time-counter, which once in a month allows the battery to charge to a 100%, no matter what the user set up at SOCmax could work. So that the battery just follows the normal hyper plan and whenever 100% is reached, afterwards it sets to the original SOCmax again. That way noone complains that expensive energy was taken from the grid or something :-D

That way, every user could decide themselves, whether to do a calibration or not.

surfer1264, 2025-04-12:
Calibration is NOT done automatically by hub1200 and 2000, Never.
The Problem with the jumps from 22 to 10 is a BMS Problem, and could not be solved externally by HA.

The only thing is:  Stop providing delivery of Energy in the House, and use of solarpower to load the battery. So only trigger/ Routine …stop discharging ….wait until 100% ….start discharging is possible

b703d0b4ecc17cd85c8f904cbbe91015, 2025-08-02:
You can simply take care of this by yourself by making an automation that sets the Maximum SOC to 100% and starts a manual charge. Once 100% is reached you trigger the automation again and set the Maximum SOC back to whatever you like (i.e. 90%). Trigger the automation every week or month or whatever interval you like and you’re done.

b703d0b4ecc17cd85c8f904cbbe91015, 2025-08-02:
You can simply take care of this by yourself by making an automation that sets the Maximum SOC to 100% and starts a manual charge. Once 100% is reached you trigger the automation again and set the Maximum SOC back to whatever you like (i.e. 90%) and the mode you want to run on (you can make it a bit nicer by writing the mode that was used before calibration to a text helper and write it back when calibration is done). Trigger the automation every week or month or whatever interval you like and you’re done.

MrSuika, 2025-08-02:
The thing with LiFePO4/Lithium iron phosphate Batteries is that the difference in voltage between empty and full is very shallow, around 3.2-3.4V, depending on if you are charging or not. Only when they get close to 0% or close to 100% the voltage goes up or down more noticeably. 2.5V is empty, 3.65V is full, that is charging current. When you let them rest, they drop down to about 3.4V at 100%.
Now the cells have 2 main ways of knowing their SoC (State of Charge) through battery voltage and through counting what goes in and what goes out, but you always have internal losses, so the counting method gets inaccurate over time, that's why you need to balance the cells every so often, especially LFP cells.
You can do a top balancing or bottom balancing, both are fine, with top balancing, you charge them until they reach their maximum charging voltage of 3.65V, with bottom balancing, you go the other way. 

But I would prefer top balancing because if your Voltage drops too low, you can damage or brick your cells, likewise, constantly keeping them at max voltage, especially in higher temps, isn't good either.

I have set my discharge limit to 5% in the summer, then they stop discharging at around 2.9V, but if you want the most cycles out of your batteries, you should set it so 10-90% or better 20-80%. Cycles till 80% remaining capacity vary from around 3000-8000 cycles, depending on the DoD (Depth of Discharge), but also on the C-rate you Charge and Discharge at. 1C means doing 1 charge cycle within an hour, so for example, charging a 2kWh battery at 2kW, 0.5 would mean half that, so charging at 1kW.
Slower charging is generally considered to be better, but newer sources say that you also want to go to the limit and do higher discharges every now and then, but that if more relevant for EVs than for home batteries, I guess.

Also, note that the voltage is on the cell level; the nominal Voltage is 3.2V, and the cells get wired up in series. With the Zendure batteries, there are 15 cells in series, giving you 48Volts, but many others are 16 cells so 51.2V.

Hope that helps you to understand the topic a little bit better.

Tl;Dr: Charge to 100% once a month or when the SoC reading gets inaccurate.

surfer1264, 2025-08-02:
https://github.com/surfer1264/ZendureWiki/wiki/Der-AB2000