Zendure/Zendure-HA #69: Integration logs ERROR

FrColin, 2025-04-11:
hello , I started install of this Integration but it give me Errors on 

Enregistreur: custom_components.zendure_ha.select
Source: custom_components/zendure_ha/select.py:55
intégration: Zendure Home assistant Integration ([documentation](https://github.com/fireson/zendure-ha), [problèmes](https://github.com/FireSon/Zendure-HA/issues))
S'est produit pour la première fois: 13:23:37 (92 occurrences)
Dernier enregistrement: 13:38:37
Error list index out of range setting state: Hyper 2000 acMode => 2

the acMode is "AC Input" if i change to "AC Output" it return immediately to "AC Input" 

And 
Enregistreur: homeassistant.util.logging
Source: util/logging.py:74
S'est produit pour la première fois: 13:37:41 (1 occurrences)
Dernier enregistrement: 13:37:41

Module custom_components.zendure_ha.zendurephase is logging too frequently. 200 messages since last count

I can't find those 200 messages

FrColin, 2025-04-11:
I think that the acmode should be as 
mapper =  { '0':'Nothing', '1':'AC input mode', '2':'AC output mode'  }

FireSon, 2025-04-11:
@FrColin can you update to version 29? There were several user having som similar error which should be fixed in this release

FrColin, 2025-04-11:
Many thanks ,

 many error logs disappear
only one new:

Enregistreur: py.warnings
Source: custom_components/zendure_ha/binary_sensor.py:12
intégration: Zendure Home assistant Integration (documentation, problèmes)
S'est produit pour la première fois: 18:44:12 (1 occurrences)
Dernier enregistrement: 18:44:12

/usr/local/lib/python3.13/site-packages/stringcase.py:247: SyntaxWarning: invalid escape sequence '\W' return re.sub("\W+", "", string)

many sensor name change , french translation thanks again

FrColin, 2025-04-11:
Since reinstalling new version 
Some new errors:

Cette erreur provient d'une intégration personnalisée

Enregistreur: custom_components.zendure_ha.zendurermanager
Source: custom_components/zendure_ha/zendurermanager.py:204
intégration: Zendure Home assistant Integration (documentation, problèmes)
S'est produit pour la première fois: 18:56:44 (8517 occurrences)
Dernier enregistrement: 23:21:21

Traceback (most recent call last): File "/config/custom_components/zendure_ha/zendurermanager.py", line 159, in on_message if device.updateProperty(key, value): ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^ File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 84, in updateProperty if not super().updateProperty(key, value): ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^ File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty if sensor.state != value: ^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state unit_of_measurement = self.unit_of_measurement ^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 499, in unit_of_measurement if (translation_key := self._unit_of_measurement_translation_key) and ( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "src/propcache/_helpers_c.pyx", line 80, in propcache._helpers_c.cached_property.__get__ File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 656, in _unit_of_measurement_translation_key raise ValueError( ...<2 lines>... ) ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
Traceback (most recent call last): File "/config/custom_components/zendure_ha/zendurermanager.py", line 167, in on_message device.updateProperty("clusterId", properties["clusterId"]) ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 84, in updateProperty if not super().updateProperty(key, value): ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^ File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty if sensor.state != value: ^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state unit_of_measurement = self.unit_of_measurement ^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 499, in unit_of_measurement if (translation_key := self._unit_of_measurement_translation_key) and ( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "src/propcache/_helpers_c.pyx", line 80, in propcache._helpers_c.cached_property.__get__ File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 656, in _unit_of_measurement_translation_key raise ValueError( ...<2 lines>... ) ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform

FireSon, 2025-04-11:
Hi sorry, it seems this error pops up when the initialization when not totally correct. I have had the same error also, but was not able to dind a cause, since it does not happen every time.
I will keep looking for it! Closing this issue for now, but let me know if something is poppoing up!

closed 2025-04-11 by FireSon

FireSon, 2025-04-12:
@FrColin well I found it!! As it turns out this was caused by the asynchronious startup of the integration, and the Zendure devices logging multiple values. I made version 31 in which it should be fixed!