Zendure/Zendure-HA #71: Hyper 2000 Temperatur

Michael6804, 2025-04-12:
Mit dem neuen Update ist die Entität Hper Temperatur nicht mehr verfügbar, vorher hat es funktioniert

SleXx88, 2025-04-12:
I just saw it, the same problem with version 1.0.30 and the SolarFlow 800.

lpetflo, 2025-04-12:
I think the sensor was (mistakenly?) renamed to Batterietemperatur (probably something like battery temperature in english), but in my case sensor.hyper_2000_hyper_tmp is still measuring the Hyper-Temp.

Kieft-C, 2025-04-12:
Check your Entitie Names, they may change up and has to reorganise.
Take care that u don't loose for the important staticstis your values.
May add template sensor's with the name of the old sensor and give them the value from the new name entities.

FireSon, 2025-04-13:
Yes, this is due to the new translation option. like @lpetflo mentioned the 'entity_id' should be the same, but the 'displayname' now is translated into the language of HA. Hopefully it will not change again :)

closed 2025-04-13 by FireSon