Zendure/Zendure-HA #74: mode CT 1.0.30 ?

sankukace, 2025-04-12:
Hello,

I just upgraded from version 1.0.28 to 1.0.30, and after an hour I’ve noticed two things so far:

CT mode no longer works for me :( :(

The battery state (sensor.xxxxx_pack_state) correctly shows charging/resting/etc. in the add-on, but Home Assistant only shows 0 or 1 instead of text :(

The most important issue is the CT mode.

Thanks in advance.

Wunni73, 2025-04-12:
und schon nach den Update, und Neustart des Systems aktiviert es sich, und Powert 800W ins Netz

Der Wert Power zeigt den Unterschied an aber nicht in Minus (-)

jeff1975, 2025-04-12:
Hallo bei Version 1.30 fehlt die AC Ladespannung

sguimbal, 2025-04-12:
> Hello,
>
> I just upgraded from version 1.0.28 to 1.0.30, and after an hour I’ve noticed two things so far:
>
> CT mode no longer works for me :( :(
>
> The battery state (sensor.xxxxx_pack_state) correctly [...]

Hello, Waht do you mean about CT ? (sorry for that)
I've a Shelly 3EM (not the pro version) and for the first time "for me at least" the Intelligent Control mode is working fine when i activate it.
With previous versions, nothing happened.
Stephane

sankukace, 2025-04-12:
I'm talking about Smart mode, sorry.

I have a Shelly 3EM and it was working perfectly with all previous versions. This must be a mistake somewhere. Of course, I set my Shelly sensor to P1 in the add-on, just like I did in the other versions.

sguimbal, 2025-04-12:
> I'm talking about Smart mode, sorry.
> 
> I have a Shelly 3proEM and it was working perfectly with all previous versions. This must be a mistake somewhere. Of course, I set my Shelly sensor to P1 in the add-on, just like I did in the other versions.

To be activated into the Official APP it could take up to 30 secondes after changing the value from Home Assistant.
Did you wait enough time ?

sankukace, 2025-04-12:
What do you mean it needs to be activated in the app?
With Fireson's add-on, I no longer needed to use the app (and my goal is precisely to stop using it).
Yes, I’ve been testing for hours, trying everything possible since I migrated to 1.0.30. A few things have issues, but the most important one for me is Smart mode.
I’ve tried with cluster mode, non-cluster mode, etc…

But I’m sure Fireson will figure it out :)

sguimbal, 2025-04-12:
> What do you mean it needs to be activated in the app? With Fireson's add-on, I no longer needed to use the app (and my goal is precisely to stop using it). Yes, I’ve been testing for hours, trying [...]

I'm using the Official APP just to confirm the order has been well transmitted to the Unit. that's it :-)

Maggi0r, 2025-04-12:
Hello, how exactly does this smart mode work?
I have a smart meter that measures my consumption in watts and also displays both positive and negative consumption. I've added this sensor to the integration. And I've disabled the bypass in the Zendure app so that it doesn't feed in more than I use. But when I do that and activate the Smart Manager, the app also shows me that smart mode is active, but nothing is fed in. I hope you can help me. I don't quite understand it. I don't have a Shelly3EM in use. Regards, Maggi

FireSon, 2025-04-12:
Hi guys, these are a lot of issues in one. It is better to seperate them in the future ;)

A lot of these issues might have to do with the problems during startup which is fixed in version 31. Can you please retest with this version?

@sankukace can you specify which sensor.xxxxx_YYYYYYY you refer to when you mean "but Home Assistant only shows 0 or 1 instead of text :("? That makes it easier to find!

@Maggi0r Is your problem the bypass? Or is the smartmode not working?

There is a big change in the way the integration handles cuircuit overload! Let's call it progressive insight ;) But when the "Device cluster" has the value: "Unknown cluster configuration, please check!!" the smartmode will NOT use this device! You have to select the correct option! This is described in the readme. This has to do with the fact that the integration also supports other Zendure devices.

FireSon, 2025-04-12:
@Maggi0r You are right, that was the whole idea I started this integration! But it is very difficult for me to know what is happening in your situation. It might be good to enable the debug logging and look what the integration is doing. (Settings/System/Logs) make sure you select "show raw logs" in that page. Enable debug logging you have to select in the xxxx/config/integrations/integration/zendure_ha page.

Did you select the proper 'device cluster' for your devices? Otherwise they will not work!

Maggi0r, 2025-04-12:
Okay, I'm attaching the logs. There are errors.
I have a Hyper. Which mode should I use? I've tried the first and second mods, but unfortunately, no change.

Logs: `ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:28.991 ERROR (paho-mqtt-client-74202:5f85482d-edad-48f2-9fd7-dd84d944bcb9) [custom_components.zendure_ha.zendurermanager] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:28.992 ERROR (paho-mqtt-client-74202:88348c0f-add9-426e-ac50-b907e0c959e7) [custom_components.zendure_ha.zendurermanager] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:28.995 ERROR (paho-mqtt-client-74202:88348c0f-add9-426e-ac50-b907e0c959e7) [custom_components.zendure_ha.zendurermanager] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zendurermanager.py", line 159, in on_message
  File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 84, in updateProperty
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty
ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:28.999 ERROR (paho-mqtt-client-74202:5f85482d-edad-48f2-9fd7-dd84d944bcb9) [custom_components.zendure_ha.zendurermanager] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zendurermanager.py", line 159, in on_message
  File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 84, in updateProperty
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty
ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:29.000 ERROR (paho-mqtt-client-74202:4cfbb137-5759-477d-b85c-3dd8c6f5de49) [custom_components.zendure_ha.zendurermanager] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:29.001 ERROR (paho-mqtt-client-74202:8b05beb7-1f63-4bad-a1d3-ec2be3ce3d84) [custom_components.zendure_ha.zendurermanager] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:29.003 ERROR (paho-mqtt-client-74202:8b05beb7-1f63-4bad-a1d3-ec2be3ce3d84) [custom_components.zendure_ha.zendurermanager] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zendurermanager.py", line 159, in on_message
  File "/config/custom_components/zendure_ha/devices/hyper2000.py", line 84, in updateProperty
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 70, in updateProperty
ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 17:12:29.005 ERROR (paho-mqtt-client-74202:c319df58-d88a-4f70-ae8d-245a8e7eaf03) [custom_components.zendure_ha.zendurermanager] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform`

Maggi

sankukace, 2025-04-12:
@FireSon 

[image] and 

[image]

n my case, it's Hyper Cuisine.
And it's in Smart mode (the mode that was available before in version 1.0.28).
Yes, I tried in a cluster, on a "dedicated circuit", ... same result :(

FireSon, 2025-04-12:
@Maggi0r the error should be fixed in version 31 did you install it?
@sankukace did you also update to version 31?

sankukace, 2025-04-12:
31 sure :)

sankukace, 2025-04-12:
Ah, all of a sudden it just started — even though I had rebooted the VM several times...
I'm looking into why, and continuing tests to see if it will restart again...
I'll keep you posted.

sankukace, 2025-04-12:
I stopped the Zendure Manager — it shut down one of my Hypers but not the second one.
I reloaded the add-on, and that shut down everything. Smart mode won't restart.

In the log I have:

2025-04-12 20:16:33.238 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 20:16:33.240 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 96, in sensorAdd
    if entity.state != value:
       ^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement
                          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 499, in unit_of_measurement
    if (translation_key := self._unit_of_measurement_translation_key) and (
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src/propcache/_helpers_c.pyx", line 80, in propcache._helpers_c.cached_property.__get__
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 656, in _unit_of_measurement_translation_key
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform

2025-04-12 20:16:33.240 ERROR (MainThread) [homeassistant.components.sensor] Platform zendure_ha does not generate unique IDs. ID Hyper Salon-outputPackPower is already used by sensor.hyper_salon_output_pack_power - ignoring sensor.Hyper Salon-output_pack_power
2025-04-12 20:16:33.240 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Entity <class 'custom_components.zendure_ha.sensor.ZendureSensor'> cannot have a translation key for unit of measurement before being added to the entity platform
2025-04-12 20:16:33.241 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 96, in sensorAdd
    if entity.state != value:
       ^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    unit_of_measurement = self.unit_of_measurement

sankukace, 2025-04-12:
Could it be an issue with Home Assistant since a recent update? Because it seems I’m seeing similar errors with other entities as well.

sankukace, 2025-04-12:
I think I’ve found the issue… I deleted an old output_pack_power entity that seemed to be duplicated, and now it seems to be working.

However, it takes longer than before — between starting Smart mode and it actually working, it now takes at least 30 seconds, whereas it used to take just a few seconds.

I’m continuing my tests, but so far it looks okay on my end.
Why that entity didn’t get removed earlier? I don’t know — but the important thing is that it seems to be working now.

sankukace, 2025-04-13:
I'm closing this thread as the Smart mode seems to be working fine. I'll open another thread for the other issues encountered.

Thanks again.

closed 2025-04-13 by sankukace