Zendure/Zendure-HA #76: new Translate de

Wunni73, 2025-04-12:
"cluster": {
"name": "Gerätecluster",
"state": {
"clusterunknown": "Unbekannte Clusterkonfiguration, bitte prüfen!!",
"clusterowncircuit": "Gerät hat einen eigenen Stromkreis oder eine eigene Phase",
"cluster800": "Cluster max. 800 W",
"cluster1200": "Cluster max. 1200 W",
"cluster2400": "Cluster max. 2400 W"
}

sankukace, 2025-04-12:
and in fr

  "cluster": {
    "name": "Regroupement des appareils",
    "state": {
      "clusterunknown": "Configuration de cluster inconnue, veuillez vérifier !!",
      "clusterowncircuit": "L’appareil a son propre circuit ou phase",
      "cluster800": "Cluster max. 800 W",
      "cluster1200": "Cluster max. 1200 W",
      "cluster2400": "Cluster max. 2400 W"

FireSon, 2025-04-12:
Thanks! Done in the next release

closed 2025-04-12 by FireSon