Zendure/Zendure-HA #79: version 31 don't work hyper 2000

Wunni73, 2025-04-12:
[image]

and it start direct after update

FireSon, 2025-04-12:
What does not work? The screenshots look nice :)

Wunni73, 2025-04-12:
smart mode is on 
hyper is in Pause  ???

in version 24 show us the P1 different to up POWER 
why was it hidden again ??

Kieft-C, 2025-04-12:
May u miss understand some Informations in the german translation?

"Ruhezustand" means, the battery does not charge oder discharge.
Not the Hyper is in Pause.

FireSon, 2025-04-13:
Do you have one Hyper2000?

Kieft-C, 2025-04-13:
I got an hyper, Jes.

tails38, 2025-04-13:
Hello, i have the .31 version and i have make the installation.
All the sensors are on unknown status.
He have detected i have 1 Hyper2000.
I use also the 2nd account with zendure cloud

FireSon, 2025-04-13:
@tails38 There was a problem with .31 and one hyper2000! I just released .32 to fix that problem, can you try it?

Wunni73, 2025-04-13:
I have 32 installed. Smart mode starts immediately? It doesn't control. The output is 0. If I turn it off and on again, it controls for 1 second, and then the output is 0. (hyper2000)

tails38, 2025-04-13:
It’s better :)

Ce message et les pièces jointes sont confidentiels et réservés à l'usage
exclusif de ses destinataires. Il peut également être protégé par le secret
professionnel. Si vous recevez ce message par erreur, merci d'en avertir
immédiatement l'expéditeur et de le détruire. L'intégrité du message ne
pouvant être assurée sur Internet, la responsabilité du groupe ne pourra
être recherchée quant au contenu de ce message. Bien que les meilleurs
efforts soient faits pour maintenir cette transmission exempte de tout
virus, l'expéditeur ne donne aucune garantie à cet égard et sa
responsabilité ne saurait être recherchée pour tout dommage résultant d'un
virus transmis.

This e-mail and the documents attached are confidential and intended solely
for the addressee; it may also be privileged. If you receive this e-mail in
error, please notify the sender immediately and destroy it. As its
integrity cannot be secured on the Internet, the Atos Origin group
liability cannot be triggered for the message content. Although the sender
endeavours to maintain a computer virus-free network, the sender does not
warrant that this transmission is virus-free and will not be liable for any
damages resulting from any virus transmitted.

closed 2025-04-16 by FireSon