Zendure/Zendure-HA #80: 1.0.31  zedure connection

EnigmuX, 2025-04-12:
since 1.0.31 everything failed 

[image]

connexion failed with the good login /passowrd !
[image]

Kieft-C, 2025-04-12:
Did u check the sensor names? they may change with the update.

EnigmuX, 2025-04-12:
all zendure sensors are unavailable because zendure connection failed

FireSon, 2025-04-13:
@EnigmuX you have to give a bit more information. What version did work? Can you enable the debug logging for the integration, and try to login again, and look at the logging if there are any clues in there?

EnigmuX, 2025-04-13:
@FireSon  version 1.0.29 worked fine, but since installing 1.0.31 everything has gone wrong. 

debug login : 

[home-assistant_zendure_ha_2025-04-13T13-13-08.764Z.log](https://github.com/user-attachments/files/19725447/home-assistant_zendure_ha_2025-04-13T13-13-08.764Z.log)

FireSon, 2025-04-13:
@EnigmuX thanks for the logfile. I have looked at it, but is seems there might be a problem with the sensor you use as a p1 meter!

```
Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/config_flow.py", line 113, in async_step_reconfigure
    user_input[CONF_P1METER] = config_entry.data[CONF_P1METER]
                               ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
KeyError: 'p1meter'

```
This means that the integration cannot read the value from the configuration. Maybe you have deleted the entry, and when you recreated it used the default value? Please check if the sensor you specified in the configuration exists and has a correct value.

EnigmuX, 2025-04-13:
@FireSon

EnigmuX, 2025-04-13:
@FireSon 

[image]

I haven't changed a thing since 1.0.29

the update has reset the configuration ?

all zendure sensors have been renamed _2 since 1.0.31

FireSon, 2025-04-13:
Yes that is possible, can you delete it and add it again? It might have to do with the enitty names...

EnigmuX, 2025-04-14:
thanks for the support 
now everything works again 

p1meter reconfigured 
new dashbord created with new sensor names

FireSon, 2025-04-14:
Great!!

closed 2025-04-14 by FireSon