Zendure/Zendure-HA #84: smart mode stop working

FrColin, 2025-04-13:
After working for few hours , the smart mode stop working.
Try to change parameter has no effect , the zendure app display smart mode but output still 0.
Stopping smart mode in zendure app will  return to on by the intégration but no output either
I notice that the smart P1 parameter is not saved in case of reconfiguration.
Making a template with the default name did not help.

Kieft-C, 2025-04-13:
Got more or less the same issue.
1.0.31

Smart mode at integration did not start.
Can’t safe the power sensor (Shelly 3em pro total sensor) 
Only “stop” (aus) works but can’t restart the smart match mode.

Should there an option to restart the mode from “intelligent mode” from zendure app? Not to integrate a new power sensor matching in HA?

FrColin, 2025-04-13:
I got change in the log:
(MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 795 from: 445 stateBatteryState.DISCHARGING

but no mqtt "iot/..." in tcpdump

FrColin, 2025-04-13:
with a 35% electricLevel 
i got this output in the log
2025-04-13 15:42:42.736 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 56 from: 404 stateBatteryState.IDLE capacity: 0 max: 0
2025-04-13 15:42:42.736 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Hyper 2000 => 0
2025-04-13 15:42:42.736 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => 0
2025-04-13 15:42:43.547 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 hyperTmp => 3061
2025-04-13 15:42:44.584 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 reverseState => 0
2025-04-13 15:42:44.584 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 acMode => 2
2025-04-13 15:42:44.584 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 lampSwitch => 0
2025-04-13 15:42:46.659 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 solarInputPower =>                                                                             217
2025-04-13 15:42:46.661 INFO (paho-mqtt-client-56519:8aa01e9e-a881-4044-a369-152a054330a6) [custom_components.zendure_ha.zenduredevice] Update Hyper 2000 outputPackPower =>                                                                             217

FrColin, 2025-04-13:
May be different scale between socSet and electricLevel in the calculation
d.capacity = d.asInt("packNum") * (d.asInt("electricLevel") - d.asInt("socMin"))
d.capacity = max(0, d.asInt("packNum") * (d.asInt("socSet") - d.asInt("electricLevel")))

why multipling by packNum  ? the electricLevel and socSet is for the global device ?
dont the capacity be equal to electricLevel ?
( may be I am wrong since i got only one hyper 200 )
[custom_components.zendure_ha.zenduredevice] Update Hyper 2000 electricLevel => 37
[custom_components.zendure_ha.zenduredevice] Update Hyper 2000 socSet => 1000

FireSon, 2025-04-13:
@FrColin The packNum is the number of batteries. In your case it does not matter, but if you have multiple devices with different number of batteries it is important to load/unload them by 'real capacity'. This is the useable part = electricLevel - socMin for discharging or socSet - electricLevel for charging, times the 'real' capacity = PackNum * 2 Kwh.
The socSet is defined for the Hyper2000 in this way
```
            self.number("socSet", "{{ value | int / 10 }}", "%", None, 5, 100, NumberMode.SLIDER),
            self.number("minSoc", "{{ value | int / 10 }}", "%", None, 5, 100, NumberMode.SLIDER),

```
which means that they are divided by then which makes in your example:
`[custom_components.zendure_ha.zenduredevice] Update Hyper 2000 socSet => 1000`
the actual socSet 100.0
Hope that explains it a bit!

But since you mentioned it, I missed a max on the discharge, which might result in a wrong calculation! Thanks for that.

FrColin, 2025-04-14:
Thanks for the explain,In my exemple i have two ab2000 and 1 hyper2000I think it could me better to take in account The soclevel of each battery not the global levelSince the battery may have different level in case of bad balance

FireSon, 2025-04-16:
Closing this issue for now

closed 2025-04-16 by FireSon