Zendure/Zendure-HA #85: State format change in sensor.hyper_pack_state after update to 1.0.31

sankukace, 2025-04-13:
Hi there,

Since the update to version 1.0.31 of the Zendure integration (previously I was on 1.0.28), I’ve noticed a strange behavior with the entity sensor.hyper_salon_pack_state.

Previously, this entity returned readable states like Repos, En charge, or Décharge (in French), which I was using directly in the power-distribution-card via the secondary_info_entity field.

However, since the update, this same entity sometimes returns numeric values like 0 or 1 instead of the expected text. This causes display issues in my dashboard (for example, the card shows 0 instead of Repos).

Here’s a summary of what I’ve observed:

The issue appeared only after updating to 1.0.31.

The history of the entity still shows the correct textual states.

But intermittently, the current state becomes a number (0, 1, etc.).

Was this change intentional in the integration, or could it be a regression?

Let me know if you need any further information (YAML config, logs, screenshots…).

Thanks a lot for your great work on the integration!

FireSon, 2025-04-13:
There was a big change in version 29 with regard to the entity_id and the way the translation of the name was handled. This could result in a change of the name of the entity. If you look at the entity inself do you see the name there?

This is how it looks for me

[image]

sankukace, 2025-04-13:
Thanks for your reply!

I just checked the entity directly in Home Assistant (sensor.hyper_salon_pack_state) using the Developer Tools.

The entity ID has not changed — it is still the same as before.

The friendly name is correct: "Hyper Salon État de la batterie"

However, the state is currently 1 (a number), whereas before it was a readable string like "En charge" (charging), "Repos" (idle), or "Décharge" (discharging).

So it seems the entity itself is now returning a numeric state instead of a translated string.

This change breaks the UI display in some cards (like power-distribution-card), which expect a readable status.

Let me know if this numeric value is expected, or if it’s a regression or localization issue introduced after version 1.0.28.

Thanks again for your support!

[image]

[image]

closed 2025-04-16 by FireSon