Zendure/Zendure-HA #86: 1.0.31 : Smart mode: only partial power delivered when one Hyper reaches low battery limit

sankukace, 2025-04-13:
Hi,

Since version 1.0.31, I’ve noticed an issue with Smart mode behavior when using two Hyper 2000 units.

When both devices are above their configured low battery limit, the total power demand is shared correctly. For example, with a 500W home consumption, both Hypers deliver power as expected.

However, when one of the two Hypers reaches its configured low battery threshold, the other one does not take over the full load.

Instead of delivering the full 500W from the remaining unit, the system only provides around 250W — as if the unavailable unit's portion was simply dropped.

Is this expected behavior for Smart mode? Or could this be a limitation or a bug introduced in the latest version?

Thanks in advance for your help and your work on the integration!

FireSon, 2025-04-13:
The expected behaviour is that the remaining device delivers 500w. I saw once that the hyper2000 stopped charging af 95% although the limit was 96%, could that be the case? Currently I just check if the limit is reached, but if the hyper2000 is not doing anything you get the observed behaviour. Maybe there is a entity which indicates that the hyper is full or empty, I did not look into that yet!

sankukace, 2025-04-13:
It seemed to work fine with version 1.0.28.

Thanks in advance :)

FireSon, 2025-04-13:
I understand, but if one of the hypers stopped delivering at 11%, and the limit is 10% the integration still 'thinks' it can deliver and assigns 250w to this device. If however the Hyper does not deliver anymore, this will lead to the error. I have observed this behaviour once, normally it stops af 10%, so it is not always. Can you check if this is really the case in your situation?

sankukace, 2025-04-13:
No. For example, with a 15% limit for Hyper1 and a 25% limit for Hyper2, and both batteries at 50%, when Hyper2 reaches 25%, it stops discharging, but Hyper1 does not take over the remaining load.

FireSon, 2025-04-14:
I think (hope) I found the cause of this problem! I have fixed it in release 33. Please reopen the issue if it still exists!

closed 2025-04-14 by FireSon