Zendure/Zendure-HA #89: 1.0.32 Smart mode :  yes and no...  and yes ... and no  :)

sankukace, 2025-04-13:
Version 1.0.32: Smart mode works when it wants to… sometimes it stops, then comes back, then stops again.
It doesn't send the surplus to a second Hyper that's on its own circuit (I also tried in cluster mode — same result).

I had to switch to the app for about 30 minutes because I had a 1000W surplus, and Smart mode should have sent power to both Hypers — but it seems it wasn’t even sending to Hyper1.

Thanks again, and good luck! :)

sankukace, 2025-04-13:
After reviewing the logs with debug mode enabled, I noticed that:

smartMode is set to 0 for both Hyper units

smartPower is also 0, even when there is solar surplus

However, in Home Assistant, the switch.hyper_salon_smart_mode is turned on

So it seems the integration is not applying Smart mode, even though the switch is active.
Could this be a bug in the current version (1.0.32)? Or a problem with sending commands to the API?

Kieft-C, 2025-04-13:
Smartmode also no work on my side. 

I can Go to "off". Where the sensor give feedback sensor.hyper_2000_auto_model_2 "kein":
before of the same sensor says " geändert zu Smart CT Leistungsregelung " but I can't came back to "smart ct mode" with choose them in the integration.

sankukace, 2025-04-14:
It seems more complicated. I had left the smart mode enabled last night even though it wasn't working, and this morning I noticed that it worked during the night, but I didn't check what time or anything... I think it's a delay or something like that, because Fireson mentioned he extended some kind of delay so that it would work for all the Zendure units, I believe.

sankukace, 2025-04-14:
I just rolled back to version 1.0.30 because otherwise the smart mode keeps turning off every few seconds, even through the Zendure app :(

FireSon, 2025-04-14:
@sankukace sorry to hear that! Unfortunately without more context it is very hard to troubleshoot the problem. I did however find an error in the power distribution which could potentionaly cause this problem as well. That is fixed in version 33.

But in order to find it I need information from the debug log and specific this type of information:
```
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: -5776 from: -4825 stateBatteryState.CHARGING capacity: 0 max: 3600
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Zen 19 => -3600 capacity 0
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Zen 19 => no action [power 0 capacity 0]
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 2 power Zen 05 => -2400 capacity 0
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Zen 66 => no action [power 0 capacity 0]
2025-04-14 12:03:03.222 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Zen 05 => no action [power 0 capacity 0]
```
This give the current production of the hous: -5776 watt. 
The division over the clusters 1 and 2 (unfortunately my batteries are full right now)
And per cluster per device the power and capacity.

Could you check in you system what these values are?

FireSon, 2025-04-14:
@sankukace Well yes and no and yes again! 

- Can it be that all you batteries are 'full'? which means electricLevel >= socSet? In that case the Smartmode will do nothing. Since there is nothing it can do. 
- Furthere more the setpoint is circling atound 0, sometimes above and sometimes below. In order to prevent too much switching to charging/discharging there is a delay that the power is 10 seconds positive or negative before it turns on
- For charging the power needs to be below -50w
-Finally, the hypers are added 'gradually' depending on the available power. Below 100w only one will work, the second will be added if the power for the second one is more than 100w (which depends on the capacity). If however it is allready active the second will be set to zero if power drops below 40w. In the logfile power never exceeds the minimum of -250 which could explain the behaviour

In this example you can see that all capacities are 0 and as a result the state = stateBatteryState.IDLE:
```
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: -157 from: -174 stateBatteryState.IDLE capacity: 0 max: 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Hyper Cuisine => -157 capacity 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper Cuisine => no action [power 0 capacity 0]
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Hyper Salon => -157 capacity 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper Salon => 0 capacity 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Hyper Cuisine => -157 capacity 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper Cuisine => no action [power 0 capacity 0]
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.zenduredevice] Update cluster 1 power Hyper Salon => -157 capacity 0
2025-04-14 15:48:55.340 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper Salon => 0 capacity 0
```

sankukace, 2025-04-14:
Maybe I’m just not logging at the right moment 😅

Today, since Smart mode wasn’t working and it was sunny, I used the Zendure app to manually send the surplus to Hyper2 (which has no PV input). Meanwhile, Hyper1 was slowly recharging using its own PVs.

This evening, I installed version 1.0.27 on a dedicated Home Assistant VM just to test it. I also disabled Smart mode on my main HA running 1.0.33.
And guess what? Version 1.0.27 is handling Smart mode properly — just like it used to!
On the other hand, 1.0.33 either does nothing, or sometimes only sends power to one Hyper (Hyper1).

I’ve disabled clustering for my Hypers in the Zendure app for weeks now, since everything was working fine with your add-on.
So in version 1.0.33, both Hypers are set as “has its own circuit or phase”, while on the VM running 1.0.27, it’s still in the original setup: “2400W per phase”.

Not sure if this helps you, but I hope so!

Anyway, I now have a backup solution so I don't lose my solar surplus: running version 1.0.27 on a separate HA instance until a future 1.0.34 or 1.0.35 fixes this 😄😄😄

Thanks again as always!

FireSon, 2025-04-14:
Well if you help me, I think we can find the problem 😄
The reason v33 does nothing is the capacity which = 0. Can you check what the electricLevel, socSet and packNum values are at that moment? The capacity is calculated in this way:

`d.capacity = max(0, d.asInt("packNum") * (d.asInt("socSet") - d.asInt("electricLevel")))`

I will add the individual values to the logging in version 34, but can you check what they are? That would give some direction to this search

sankukace, 2025-04-14:
electricLevel for Hyper1: 98
electricLevel for Hyper2: 92

socSet for both Hyper1 and Hyper2: 100
minSoc for both: 50

packNum for Hyper1: 2
packNum for Hyper2: 2

sankukace, 2025-04-14:
One strange difference between v27 and v33:

I no longer have a model for the Zendure Manager.

[image]

FireSon, 2025-04-15:
@sankukace Just updated the version, found the issue with the starting of the smart mode. Tested it today and it seems to work ;) but then again, it mostly does if I use it myself 😄

sankukace, 2025-04-15:
Cool, fingers crossed — I’m going to test it right away :)

Thanks a lot!

sankukace, 2025-04-15:
It still doesn’t seem to work, but there’s still a bit of sunlight left, so I’ll wait a bit...
Otherwise, I’ll run version 1.0.27 this evening as a backup :)

By the way, is it normal that the Zendure Manager shows as “unknown”? (like in the screenshot above)

sankukace, 2025-04-16:
While searching, I found some old entities that I just deleted, and Smart Mode just started working... I hope it holds :) I'm keeping a close eye on it and will let you know so we can close the thread.
With all these entities that have changed :) ...
Thanks again!

sankukace, 2025-04-16:
Well, after several hours, it’s working, but not at all like it did before. It charges when there isn’t enough surplus — or the opposite... I’m closing the thread and will wait a few days to try to understand things better (and follow the updates).

Thanks again, and keep it up… especially with local support :)

closed 2025-04-16 by sankukace