Zendure/Zendure-HA #95: v33

Wunni73, 2025-04-14:
[image]
it send data
Hyper doesn't Work

[image]

Kieft-C, 2025-04-14:
Did u checked the entities names? With some updates & language package the sensor names changes sadly up.

wick3da, 2025-04-14:
Ah sorry - there are new names for the displayed entities after all. Cool, then I'll quickly change all 30 of them... But why is the communication between the hyper and shelly suddenly not working anymore? Thanks for help!

wick3da, 2025-04-14:
After redownloading the integration and re-adding the entities (and manually deleting the old, unusable ones), it seems to be working normally again now. I hope that the naming of the entities won't change anymore in the future ;-) Otherwise, it's a really top-notch integration!

FireSon, 2025-04-15:
@wick3da There no plans to do that :-)

closed 2025-04-16 by FireSon