Zendure/Zendure-HA #98: v 34 a little better

Wunni73, 2025-04-15:
It's regulating now.
But when it's +, it hits with 800W; when it's -, the regulation is fine.

praise for your efforts

Wunni73, 2025-04-15:
I found out when it powers in with 800W, when it switches off. It only takes a few seconds, then it is back in Smart Mode and regulates

Wunni73, 2025-04-15:
The current ToDo is that when it's off the power is 0W, not 800W now.

FireSon, 2025-04-15:
Is your outputlimit 800w? If the zendure manager is off, the hyper2000 returns to normal operation, which means depending on the AC Mode, charging/discharging with the inpu/output limit.

Wunni73, 2025-04-15:
yes my limit is 800W.
yes the Zendure manager is off.
and the auto_model is 0  None
and powert 800W out

Wunni73, 2025-04-15:
I take ac_mode to input ,and  input = 0 
and it works 👍 :-)

Wunni73, 2025-04-16:
it works over Night perfekt

Is it normal that it charges from Grid ??

[image]

closed 2025-04-18 by Wunni73