Zendure/Zendure-HA #101: V1.0.34 Smart mode solar bias or offset

faxenclown120d, 2025-04-16:
I checked my energy stats today and noticed, that 
1) the smart mode in the HACS worked pretty fine until dawn/sunrise
2) the smart mode didn´t work at higher level of sunlight (solar energy > grid consumption? )and when i started it, my hyper stopped working and pushed 100% solar energy into the battery, while all electricity came from the grid
3) in the evening, the situation normalized and i realized, that the energy taken from the grid is exactly the remaining solar input.

Here some visualisation:
Pic 1) 
Phase A Energy taken from the grid - until 5AM hust 7Wh loss - that´s phantastic!
Phase B are some hickups, caused by Shelly Plug covered devices, where the power- adjustment didn´t work (typically in smart mode it did until i installed the Zendure-HA yesterday - no offense! I can live with manual adjustments at day and smart mode at night)
Phase C grid consumption and solar input ran parallel
[image]

Pic 2) 
At 7PM i switched the smart mode on for some minutes, but realized, that grid consumption was about 120W, while solar input was about 90W. 40 minutes later (almost sunset) i switched the smart mode on again and realized, that grid consumption and solar input run parallel (again) ... so i think, if you subtract the solar input somewhere (my programming skills are fairy limited - but i´m a good data analyst ;) ), the smart mode will start to work better

[image]

FireSon, 2025-04-16:
Thanks for the great explanation! Indead a great data analyst 😄

The problem at this moment is that I have not yet connected my solarpanels to the hyper2000. I plan to do that next week. Once I have them up and running, it will be much easier to investigate and add your suggestion.

harrymayr, 2025-04-16:
I have a similar problem with the smart mode. My configuration is a little different. I use the Hyper 2000 without any panels directly connected. So I use the AC-mode to load the battery from my large PV. I also have no shelly to get my Input/Output from the grid, but also an optical sensor, which generates the _Smart Meter Leistung_ (used for the Zendure HA) data and from that the _Power Sent_ (into grid) and _Power Add_ (from grid) data.
If there is enough solar power and the power sent into grid is larger than 1200W, everything is OK (until 11:37). Then I started to load my car, still enough solar power, but the power which is sent to the grid drops below 1200W (around 700W). The smart mode then also dropped the charging power of the battery to the same range, even there was enough solar power to load more.
Around 12:00 some clouds came up and additional power was used for cooking. Here you can see (12:08), that if the power from the grid exceeds the solar power, the smart mode set the charging power to maximum.

So for now, I will set input and output manually. Also I assume that the time-tables in the smart mode prevent the Hyper to switch to Output mode.

[image]

closed 2025-05-05 by FireSon